openwch/arduino_core_ch32 #149: Serial does not work on CH32L103

21km43, 2024-09-24:
Serial object does not work on CH32L103

```
void setup() {
  Serial1.begin(115200);
}

void loop() {
  Serial1.println("Hello");
  delay(1000);
}
```

maxint-rd, 2024-09-26:
Have you tried using Serial.begin instead of Serial1.begin?

21km43, 2024-09-26:
Yes, and the result is same

21km43, 2024-10-01:
After install board via package_ch32v_index.json, overwrite "%HOMEPATH%\AppData\Local\Arduino15\packages\WCH\hardware\ch32v\1.0.4" to main repository.

ex:
```
cd "%HOMEPATH%\AppData\Local\Arduino15\packages\WCH\hardware\ch32v"
rd /s 1.0.4
git clone https://github.com/openwch/arduino_core_ch32.git
rename arduino_core_ch32 1.0.4
```

shanghuo, 2024-10-02:
I can indeed see the board after this operation, but uploading the program is invalid. It displays "compilation successful" and "upload completed", but the program on the board is still the test program I burned with "MounRiver Studio" before, not the test program I wrote with "Arduino IDE"
我这样操作后确实可以看到板子，但是上传程序是无效的，它显示“编译成功”和“上传完成”，但是板子上的程序依旧是我之前用“MounRiver Studio”烧录的测试程序，而不是我使用“Arduino IDE”编写的测试程序
I don't know if your program just cannot use serial port or if you have encountered the same situation as me. Is there a solution?
不知您的程序只是不能使用串口还是和我遇到了一样的情况，有解决方案么？
Here is my test code. I connected an LED to PB5 and currently have the serial cable connected to PA9 and PA10
以下是我的测试代码，PB5我接了一个LED，串口线我目前接在PA9和PA10
```
void setup() {
  Serial.begin(115200);
  pinMode(PB5, OUTPUT);
}

void loop() {
  Serial.println("Hello");
  digitalWrite(PB5, HIGH);
  delay(1000);
  digitalWrite(PB5, LOW);
  delay(1000);
}
```

shanghuo, 2024-10-02:
Now I have discovered some situations, the specific process can be found in [issues-155](https://github.com/openwch/arduino_core_ch32/issues/155).

Please forgive my poor English. In order to express the problem I encountered more fully, I have chosen to ask in my native language, which is Chinese. I apologize for any inconvenience caused and hope you can understand. Thank you very much

## PB5引脚无效

目前我将pb1和pb5都连接了led，发现以下代码可以运行
```
void setup() {
  pinMode(PB1, OUTPUT);
}

void loop() {
  digitalWrite(PB1, HIGH);
  delay(1000);
  digitalWrite(PB1, LOW);
  delay(1000);
}
```
以下代码不能实现控制led，但在运行
```
void setup() {
  pinMode(PB5, OUTPUT);
}

void loop() {
  digitalWrite(PB5, HIGH);
  delay(1000);
  digitalWrite(PB5, LOW);
  delay(1000);
}
```
因为我尝试了下面这样的代码，并变换了接线方式测试了
```
void setup() {
  //Serial.begin(115200);
  pinMode(PB1, OUTPUT);
  pinMode(PB5, OUTPUT);
}

void loop() {
  //Serial.println("Hello");
  digitalWrite(PB1, HIGH);
  digitalWrite(PB5, HIGH);
  delay(1000);
  digitalWrite(PB1, LOW);
  digitalWrite(PB5, LOW);
  delay(1000);
}
```
也就是说PB5引脚是无效的。

## PB6引脚未定义
同样的，我还试了PB6，发现PB6引脚编译时直接报错

## Serial和Serial1无效
正如这个issue所说，Serial和Serial1都是没有效果的，我怀疑和PB5引脚可能有相似的问题。而Serial2和Serial3的情况又类似于PB6，编译时说未定义，理论上这个芯片是有这些串口的。

## PA9与PA10点灯及后续测试
1. 之后，我突然想到，会不会是PA9或PA10有什么问题，然后我进行了点灯测试，发现PA9和PA10都可以正常点灯。
2. 然后我测试了这段代码
```
void setup() {
  Serial1.begin(115200);
  pinMode(PB1, OUTPUT);
}

void loop() {
  digitalWrite(PB1, HIGH);
  delay(1000);
  digitalWrite(PB1, LOW);
  delay(1000);
  digitalWrite(PB1, HIGH);
  delay(1000);
  digitalWrite(PB1, LOW);
  delay(1000);
  Serial1.println("Hello");
  digitalWrite(PB1, HIGH);
  delay(1000);
  digitalWrite(PB1, LOW);
  delay(1000);
  digitalWrite(PB1, HIGH);
  delay(1000);
  digitalWrite(PB1, LOW);
  delay(1000);
}
```
发现`Serial1.begin(115200);`并没卡住程序，但`Serial1.println("Hello");`会卡住程序，导致后续代码无法执行，好像是死机了。
3. 再后来，我尝试了arduino的SoftwareSerial.h，发现对于ch32l103来说，这个头文件是没有的，我还跟着报错的猜测性提示尝试了esp32的HardwareSerial的写法，也是不行的。至此，我的尝试结束了。
4. 此外，我还发现了一个存储库，那个[存储库](https://github.com/Community-PIO-CH32V/platform-ch32v)的这块板(genericCH32L103C8T6)的实现也有这些问题

21km43, 2025-01-10:
Sorry to delay for response and thanks for testing some cases.

I tried these code and checked that this program halts when executing 'Serial1.println("Hello");'. 

Oddly enough, this program does not stop when `system/CH32L10x` directory replaced to latest source code (from MounRiver Studio's SDK). But it stills no output to Serial monitor.

I'll continue to check the reason of this bug.

> Now I have discovered some situations, the specific process can be found in [issues-155](https://github.com/openwch/arduino_core_ch32/issues/155).
> 
> Please forgive my poor English. In order to express the problem I encountered more fully, I have chosen to ask in my native language, which is Chinese. I apologize for any inconvenience caused and hope you can understand. Thank you very much
> 
> 在[issues-155](https://github.com/openwch/arduino_core_ch32/issues/155)中，我解决了我自己遇到的无法上传的问题。现在遇到的问题应该与这个issues的一致，甚至更多。具体如下
> ## PB5引脚无效
> 
> 目前我将pb1和pb5都连接了led，发现以下代码可以运行
> 
> ```
> void setup() {
>   pinMode(PB1, OUTPUT);
> }
> 
> void loop() {
>   digitalWrite(PB1, HIGH);
>   delay(1000);
>   digitalWrite(PB1, LOW);
>   delay(1000);
> }
> ```
> 
> 以下代码不能实现控制led，但在运行
> 
> ```
> void setup() {
>   pinMode(PB5, OUTPUT);
> }
> 
> void loop() {
>   digitalWrite(PB5, HIGH);
>   delay(1000);
>   digitalWrite(PB5, LOW);
>   delay(1000);
> }
> ```
> 
> 因为我尝试了下面这样的代码，并变换了接线方式测试了
> 
> ```
> void setup() {
>   //Serial.begin(115200);
>   pinMode(PB1, OUTPUT);
>   pinMode(PB5, OUTPUT);
> }
> 
> void loop() {
>   //Serial.println("Hello");
>   digitalWrite(PB1, HIGH);
>   digitalWrite(PB5, HIGH);
>   delay(1000);
>   digitalWrite(PB1, LOW);
>   digitalWrite(PB5, LOW);
>   delay(1000);
> }
> ```
> 
> 也就是说PB5引脚是无效的。
> ## PB6引脚未定义
> 
> 同样的，我还试了PB6，发现PB6引脚编译时直接报错
> ## Serial和Serial1无效
> 
> 正如这个issue所说，Serial和Serial1都是没有效果的，我怀疑和PB5引脚可能有相似的问题。而Serial2和Serial3的情况又类似于PB6，编译时说未定义，理论上这个芯片是有这些串口的。
> ## PA9与PA10点灯及后续测试
> 
>     1. 之后，我突然想到，会不会是PA9或PA10有什么问题，然后我进行了点灯测试，发现PA9和PA10都可以正常点灯。
> 
>     2. 然后我测试了这段代码
> 
> 
> ```
> void setup() {
>   Serial1.begin(115200);
>   pinMode(PB1, OUTPUT);
> }
> 
> void loop() {
>   digitalWrite(PB1, HIGH);
>   delay(1000);
>   digitalWrite(PB1, LOW);
>   delay(1000);
>   digitalWrite(PB1, HIGH);
>   delay(1000);
>   digitalWrite(PB1, LOW);
>   delay(1000);
>   Serial1.println("Hello");
>   digitalWrite(PB1, HIGH);
>   delay(1000);
>   digitalWrite(PB1, LOW);
>   delay(1000);
>   digitalWrite(PB1, HIGH);
>   delay(1000);
>   digitalWrite(PB1, LOW);
>   delay(1000);
> }
> ```
> 
> 发现`Serial1.begin(115200);`并没卡住程序，但`Serial1.println("Hello");`会卡住程序，导致后续代码无法执行，好像是死机了。 3. 再后来，我尝试了arduino的SoftwareSerial.h，发现对于ch32l103来说，这个头文件是没有的，我还跟着报错的猜测性提示尝试了esp32的HardwareSerial的写法，也是不行的。至此，我的尝试结束了。 4. 此外，我还发现了一个存储库，那个[存储库](https://github.com/Community-PIO-CH32V/platform-ch32v)的这块板(genericCH32L103C8T6)的实现也有这些问题
> 
> @21km43 I'm very sorry to disturb you. Thank you for your help before. May I ask if you can also reproduce the problem I mentioned here.

maxint-rd, 2025-01-10:
For what it's worth:  in the past I've had some great success fixing code on the CH32V003 and CH32X033 by using the real time debugger in the Arduino IDE 2.x. It allowed me to set breakpoints and watch variable values to get to the base of the issue and along the way get better understanding of the code. Have you tried using the debugger yet?

21km43, 2025-01-10:
I have not yet debugged using the real-time debugger and will do so in the future soon.