openwch/arduino_core_ch32 #150: Trouble Debugging CH32V203C8T6 Using Arduino

wburchard, 2024-09-28:
The marked sentence in the below image leads me to think that Arduino should be able to debug code on CH32V boards, but Arduino seems to not be able to do this.

[image]

Am I misunderstanding this sentence?  If so, could you please direct me to a description of how to debug code on these chips?  

Thank you!

maxint-rd, 2024-09-28:
In the past months I've been able to use the debugger within Arduino 2.3.2 IDE (in Windows 11) to debug the CH32V003 quite successfully.
At the moment I don't have the CH32V203, but coincidentally earlier today I ordered some of these chips. Hopefully within a few weeks I can upload some code and then let you know my experience.

In the mean time: 
What is your environment? 
Do you perhaps also have the V003 to try first?

wburchard, 2024-09-29:
I'm running Arduino 2.3.3 on Windows 11.  I just hooked up a CH32V003F4P6.
I followed the instructions found at  GitHub - openwch/arduino_core_ch32:
Core library for CH32duino <https://github.com/openwch/arduino_core_ch32>.
While I can build and flash sketches, Arduino tells me that debugging is
not supported:

[image: image.png]

maxint-rd, 2024-09-29:
If debugging isn't working for the CH32V003 either, you may not have the latest version of the CH32 Arduino core installed. 

Please note that the latest published release is 1.0.4 (dated december 2023), but many new features that have been added are not in that release. For those features you need the code in the main branche. 

Debugging was merged in PR #68 "Add debug feature" in february 2024. In the [last comment](https://github.com/openwch/arduino_core_ch32/pull/68#issuecomment-2047761549) of that PR you can read what someone else did to get debugging working.

(BTW. This core is maintained by @TianpeiLee in his spare time. Although it would be nice to have the latest code published into a proper release, this is the better than no support at all).

wburchard, 2024-09-29:
Ah, makes sense now.  Could you possibly point me to instructions on how to
build and use the latest CH32 Arduino core?

maxint-rd, 2024-09-29:
> Could you possibly point me to instructions on how to build and use the latest CH32 Arduino core?

I installed release 1.0.4 using the library manager and then copied over the newer sources. Read [this post](https://github.com/openwch/arduino_core_ch32/pull/68#issuecomment-2047761549) for more details.

In my version I've also made some other changes, for instance to have [interrupt driven Serial1](https://github.com/openwch/arduino_core_ch32/pull/145) and be able to [read VCC](https://github.com/maxint-rd/arduino_core_ch32/tree/Fix-reading-PADC_VREF).  For my CH32V003 projects I also modified the board.txt to allow enabling modules in the IDE. For using watchdog and sleep I modified [Adafruit_SleepyDog](https://github.com/maxint-rd/Adafruit_SleepyDog).

wburchard, 2024-10-01:
That did the trick!  I'm now successfully debugging my CH32V board.

I did run into the issue where the IDE hangs while downloading index
files.  However, after finding that others have encountered this and
reading the comment chain, I found by unplugging and replugging the board
that issue disappeared.

Thanks for your help!  I really appreciate it!

maxint-rd, 2024-10-01:
Good to see you got it working! When I got debugging working I was very pleased.
It allowed me to implement features that would otherwise be very hard, such as my PR for I2C-slave and interrupt driven Serial1 to fix lost data in Serial.read().

Congratulations and happy debugging!

closed 2025-04-27 by yangliangquan