openwch/arduino_core_ch32 #151: GPIO PC3 issue on Arduino CH32X035

MostafaKhan1971, 2024-09-29:
Hi,
I have been struggling with this for past couple of hours and could not find any obvious issues with library files.
When using the pin PC3/PIN_A13/A13 (all refer to the same GPIO Pin) in digital write or read or analog read function, the result I get is to and from pin PA1. But if I use the pin number 20 for digital read and write then it works fine and sends or receives signals to and from PC3. But analog read always reads 0 if pin number 20 is used. FYI, I am working with Arduino.

Thanks

maxint-rd, 2024-09-29:
Do you have the ADC enabled?  Have you tried using value A13 instead of the number 20? Is your clock setting correct? (In issue #141 and #94 people got zeros when their ADC was turned off. In issue #110 someone had ADC issues due to incorrect clock settings). 

Currently I only have CH32V003 chips. In my fan controller project I use A0 (PA2) to measure temperature. Recently I ordered some CH32X033 chips. These are similar to the X035, but in a TSSOP20 package. Once I have those I may be able to do some more relevant testing...

taisirou, 2024-09-30:
Is it the same as what I wrote?
[https://github.com/openwch/arduino_core_ch32/issues/95#issue-2302210063](https://github.com/openwch/arduino_core_ch32/issues/95#issue-2302210063)

MostafaKhan1971, 2024-09-30:
> Do you have the ADC enabled? Have you tried using value A13 instead of the number 20? Is your clock setting correct?

ADC is enabled. I did try PC3/PIN_A13/A13 all of these combinations.
Not sure about clock settings as I haven't modified anything about it. And clock doesn't matter much in this case because I can read a range of 12 bit values using A13 but reading comes from PA1 instead of PC3/A13. ADC reading from other pins also work correctly.

MostafaKhan1971, 2024-09-30:
> Is it the same as what I wrote? [#95 (comment)](https://github.com/openwch/arduino_core_ch32/issues/95#issue-2302210063)

Yes. I haven't noticed it. Should I close this thread and move my query to there?

taisirou, 2024-09-30:
> Yes. I haven't noticed it. Should I close this thread and move my query to there?
Either is fine.

maxint-rd, 2024-09-30:
> ADC reading from other pins also work correctly.

Okay, in that case it sounds like similar to issue #95 . Perhaps indeed add something to that thread. When I receive my CH32X033 chips I will do some similar tests and add if I find issues.

MostafaKhan1971, 2024-09-30:
I have moved this issue to issue #95

MostafaKhan1971, 2024-09-30:
I have moved this issue to issue #

closed 2024-09-30 by MostafaKhan1971