openwch/arduino_core_ch32 #153: Question: Where is the correct assignment of ADC1_IN7 in CH32V003F4P6?

zer0ohm, 2024-09-30:
Hello.
I'm currently designing an application board (hobby use) using CH32V003F4P6.
I have researched three documents to verify the pin assignments:

* WCH's datasheet - https://www.wch.cn/downloads/CH32V003DS0_PDF.html
* variant_CH32V003F4.cpp
* PeripheralPins.c

A question arose here. In the datasheet and variant_CH32V003F4.cpp, ADC1_IN7(A7) is assigned to PD4. However, PeripheralPins.c lists it as PA4.
(Note: According to the datasheet, only PA1 and PA2 exist for PORT-A.)

Can you please tell me if this is just a typo or if it is not used in the specification?

Thank you.

maxint-rd, 2024-09-30:
Looking at the [pinout diagram](https://github.com/user-attachments/files/17190837/CH32V003_pinouts.pdf) of the F4P6,
[image]
I  think this is a mistake in [PeripheralPins.c](https://github.com/openwch/arduino_core_ch32/blob/main/variants/CH32V00x/CH32V003F4/PeripheralPins.c). I think the pin mapping of both A6 and A7 are incorrect and should be PD_4 and PD_6, like so:
```
{PD_6,      ADC1, CH_PIN_DATA_EXT(CH_MODE_INPUT, CH_CNF_INPUT_ANALOG, 0, AFIO_NONE, 6)}, // ADC1_IN6
{PD_4,      ADC1, CH_PIN_DATA_EXT(CH_MODE_INPUT, CH_CNF_INPUT_ANALOG, 0, AFIO_NONE, 7)}, // ADC1_IN7
```

I haven't done any ADC tests on those pins yet, but did use digitalWrite to know for sure that on the TSSOP-20 pin 1 is PD4. I also believe there are no pins PA4 or PA6 on the F4P6.

zer0ohm, 2024-09-30:
Thank you for your response.
> I think this is a mistake in [PeripheralPins.c](https://github.com/openwch/arduino_core_ch32/blob/main/variants/CH32V00x/CH32V003F4/PeripheralPins.c). I think the pin mapping of both A6 and A7 are incorrect and should be PD_4 and PD_6, like so:

I was not aware of PD_6 mistake....

> I haven't done any ADC tests on those pins yet, but did use digitalWrite to know for sure that on the TSSOP-20 pin 1 is PD4.

I will test the ADC once the prototype-PCB is complete.

zer0ohm, 2024-12-11:
Sorry delay for test.
I modified PA_4 in PeripheralPins.c to PD_4 and then ran analogRead(PD4).

```
uint8_t ADC_pin_array[] = { PA2, PA1, PC4, PD2, PD3, PD4 }; // A0~A7. PD5 and PD6 using by UART.
size_t ADC_array_Count = ( sizeof(ADC_pin_array) / sizeof(ADC_pin_array[0]) );

void setup() {
  // PinMode - INPUT
  for (size_t pmd_cnt = 0; pmd_cnt < ADC_array_Count + 1; pmd_cnt++) {
    pinMode(ADC_pin_array[pmd_cnt], INPUT_ANALOG);
  }

  Serial.begin(115200);
  // Leonardo: wait for serial port to connect
  while ( !Serial ) { /* none */ }

  while (1) {
      Serial.print("adc_pin: ADC_pin_array[");
      Serial.print(ADC_array_Count - 1, DEC);
      Serial.print("], ");
      Serial.print("adc_value:");
      uint16_t adc_pin = analogRead(ADC_pin_array[ADC_array_Count - 1]);
      Serial.println(adc_pin, DEC);
      delay(1000);
  }  
} // end of setup

void loop() { /* none */ }
```

[image]
[image]

The extremely low ADC value read is, I guess depend circuit(see attach schematic pic)...  So, I don't think it's a problem as far as the function of the chip.
[image]

maxint-rd, 2025-05-06:
See also PR #160