openwch/arduino_core_ch32 #155: ch32l103c8t6无法上传

shanghuo, 2024-10-02:
## ch32l103c8t6无法上传

我使用在taobao“WCH沁恒微电子官方店”购买的评估板套件“CH32L103C8T6-LinkE套件一”中的CH32L103C8T6评估板和WCH-LinkE，用“MounRiver Studio”上传测试程序运行正常，而我用“Arduino IDE”使用本库时，编写的测试代码在点击“Arduino IDE”的上传按钮后，虽然显示“编译成功”和“上传完成”，但板上运行的代码依旧是之前用“MounRiver Studio”上传的测试代码，而不是“Arduino IDE”中编写的代码。我的测试代码如下：
```
void setup() {
  Serial.begin(115200);
  pinMode(PB5, OUTPUT);
}

void loop() {
  Serial.println("Hello");
  digitalWrite(PB5, HIGH);
  delay(1000);
  digitalWrite(PB5, LOW);
  delay(1000);
}
```
它确实编译成功，但完全没有上传进板子中。
我安装的“Arduino IDE”版本为2.3.3，“[arduino_core_ch32](https://github.com/openwch/arduino_core_ch32)”是昨天通过git按[issues-149](https://github.com/openwch/arduino_core_ch32/issues/149)中的方式克隆的。

## 另外，我还观察到：

1. 
当我Upload method选择WCH-SWD时，上传步骤执行到这一步时：
```
"C:\Users\Admin\AppData\Local\Arduino15\packages\WCH\tools\openocd\1.0.0/bin/openocd" -f "C:\Users\Admin\AppData\Local\Arduino15\packages\WCH\tools\openocd\1.0.0/bin/wch-riscv.cfg" -c init -c halt -c "program {C:\Users\Admin\AppData\Local\Temp\arduino\sketches\ABED840653BE2F82BBEF71054EE31C8E/ch32l103_test.ino.elf} verify; wlink_reset_resume; exit;"
```
报错如下：
```
Open On-Chip Debugger 0.11.0+dev-02415-gfad123a16-dirty (2023-05-05-13:43)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'sdi'
Warn : Transport "sdi" was already selected
Ready for Remote Connections
Info : WCH-LinkE  mode:RV version 2.12 
Error:  communication fail,please contact [support@mounriver.com]
```
我确定我这时是关闭了“MounRiver Studio”，只运行着当前这个“Arduino IDE”的，且WCH-LinkE(包括它的串口)都没有被占用。

2.
当我Upload method选择WCH-ISP时，上传步骤执行到这一步时：
```
"{runtime.tools.wchisp.path}/wchisp"  flash "C:\Users\Admin\AppData\Local\Temp\arduino\sketches\ABED840653BE2F82BBEF71054EE31C8E/ch32l103_test.ino.elf"
```
没有任何输出，直接瞬间提示“上传完成”。

## 我的接线：
1. 电脑通过USB连接`WCH-LinkE`
2. `WCH-LinkE`与`CH32L103C8T6评估板`接线如下表：
WCH-LinkE | CH32L103C8T6评估板
---|---
RX | PA9(tx1)
TX | PA10(rx1)
TDO | 悬空NC
TDI | 悬空NC
RST | NRST
SWCLK/TCK | PA14
SWDIO/TMS | PA13
GND | GND
3V3 | VCC
5V | VBUS
3. 为了测试上方代码，我还将`CH32L103C8T6评估板`的`PB5`与`LED2`连接（我通过观察测试发现`LED2`引脚是低电平点亮，但现在用“Arduino IDE”编程时，因为我给观察到程序虽然显示“上传完成”但实际并没有烧录进去，所以这个LED也并没有效果）

Please forgive my poor English. In order to express the problem I encountered more fully, I have chosen to ask in my native language, which is Chinese. I apologize for any inconvenience caused and hope you can understand. Thank you very much

shanghuo, 2024-10-02:
补充：
我重启电脑后，没有给“Arduino IDE”管理员权限时，当我Upload method选择WCH-SWD，在执行到这一步时：
```
Sketch uses 8172 bytes (12%) of program storage space. Maximum is 65536 bytes.
Global variables use 492 bytes (2%) of dynamic memory, leaving 19988 bytes for local variables. Maximum is 20480 bytes.
"C:\Users\Admin\AppData\Local\Arduino15\packages\WCH\tools\openocd\1.0.0/bin/openocd" -f "C:\Users\Admin\AppData\Local\Arduino15\packages\WCH\tools\openocd\1.0.0/bin/wch-riscv.cfg" -c init -c halt -c "program {C:\Users\Admin\AppData\Local\Temp\arduino\sketches\ABED840653BE2F82BBEF71054EE31C8E/ch32l103_test.ino.elf} verify; wlink_reset_resume; exit;"
```
也变为了没有任何输出，直接提示“上传完成”。

不过给“Arduino IDE”管理员权限后，依旧报错
```
Open On-Chip Debugger 0.11.0+dev-02415-gfad123a16-dirty (2023-05-05-13:43)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'sdi'
Warn : Transport "sdi" was already selected
Ready for Remote Connections
Info : WCH-LinkE  mode:RV version 2.12 
Error:  communication fail,please contact [support@mounriver.com]
```

我电脑系统是win11专业版22H2（22621.4169，Windows Feature Experience Pack 1000.22700.1034.0）

maxint-rd, 2024-10-02:
> Error:  communication fail,please contact [support@mounriver.com]

Compilation may be succesfull, but upload failed to communicate. I suggest to use the WCH link utility to see if it can properly control the chip (read data in RV mode, reset, etc). Perhaps also clear the entire flash and try again?

shanghuo, 2024-10-02:
我使用C:\MounRiver\MounRiver_Studio\ExTool\SWDTool目录下的WCH-LinkUtility手动烧录用MounRiver Studio生成的hex文件，烧录后可以正常运行MounRiver Studio中我编写的代码，而我用WCH-LinkUtility手动烧录Arduino IDE生成的ch32l103_test.ino.hex文件，烧录后并没有任何反应。
不过这样烧录后，明显ch32l103_test.ino.hex烧录进去了，因为是没有任何反应了，而不是像之前Arduino IDE中点上传时的，虽然写的“上传完成”但板子上的MounRiver Studio的程序还能运行，而Arduino IDE的程序并没进去。

我使用C:\MounRiver\MounRiver_Studio\ExTool\WCHISPTool的WchIspStudio.exe也是和使用WCH-LinkUtility一样的效果（虽然这样的话需要烧录时将boot0拉高）

shanghuo, 2024-10-02:
> > 错误：通信失败，请联系 [[support@mounriver.com](mailto:support@mounriver.com)]
> 
> 编译可能成功，但上传无法通信。我建议使用 WCH link 实用程序，看看它是否能正确控制芯片（在 RV 模式下读取数据、重置等）。也许也可以清除整个闪光灯并重试？

This can burn the "ch32l103_test. ino. hex" file generated by the Arduino IDE into it, but it doesn't seem to run effectively. I tried the following codes but none of them worked. The hex file generated by burning "MounRiver Studio" can run.
这可以将Arduino IDE生成的“ch32l103_test.ino.hex”文件刻录到其中，但似乎无法有效运行。我尝试了以下代码，但都不起作用。刻录“MounRiver Studio”生成的十六进制文件可以运行。
```
void setup() {
  Serial.begin(115200);
}

void loop() {
  Serial.println("Hello");
}
```
and
```
void setup() {
  Serial1.begin(115200);
}

void loop() {
  Serial1.println("Hello");
}
```
and
```
void setup() {
  pinMode(PB5, OUTPUT);
}

void loop() {
  digitalWrite(PB5, HIGH);
  delay(1000);
  digitalWrite(PB5, LOW);
  delay(1000);
}
```

maxint-rd, 2024-10-02:
Your code seems okay.

> This can burn the "ch32l103_test. ino. hex" file generated by the Arduino IDE into it, but it doesn't seem to run effectively. 

Unfortunately I don't have the CH32L103 chip, so I can't test. Hopefully someone else can help; Good luck!

shanghuo, 2024-10-02:
> Your code seems okay.
> 
> > This can burn the "ch32l103_test. ino. hex" file generated by the Arduino IDE into it, but it doesn't seem to run effectively.
> 
> Unfortunately I don't have the CH32L103 chip, so I can't test. Hopefully someone else can help; Good luck!

It's okay, thank you for your answer. It has been a great help to me in troubleshooting the problem

我用`C:\Users\Admin\.platformio\packages\tool-openocd-riscv-wch\bin`替换了`C:\Users\Admin\AppData\Local\Arduino15\packages\WCH\tools\openocd\1.0.0\bin`，现在可以上传成功了，看来是openocd版本的问题，不过我现在遇到了更多的问题，比如[issues-149](https://github.com/openwch/arduino_core_ch32/issues/149)，当然还有更多与之相关的问题。接下来我会关闭这个issues，并在[issues-149](https://github.com/openwch/arduino_core_ch32/issues/149)中补充后续问题。

closed 2024-10-02 by shanghuo