openwch/arduino_core_ch32 #157: CH32V003J4M6: SoftwareSerial() does not work and the UART is not available.

blopa1961, 2024-10-08:
I know this is not a help forum, but since there's no information anywhere regarding my problem I'm posting it here; maybe someone can show some light on this.
I'm trying to convert my project:
https://www.instructables.com/An-SPORT-Compass-for-FrSky-Radio-Control-Telemetry/
to the CH32V003
The project is a simple protocol converter/interface which reads an I2C magnetometer and transmits the values using a serial port.
The SOP8 version of the CH32V003 is supposed to have both a UART and I2C interfaces available; however, since my design has a crystal (necessary for the UART time base), it seems the serial port TX pin (PD6) is in use by the crystal.
My connections are:
PIN  Function
1 OSCI (24MHz crystal)
2 GND
3 OSCO (24MHz crystal)
4 VCC 5V
5 SDA
6 SCL
8 RX (URX?)
So far so good, but pin 7 is NOT TX; how am I supposed to connect the UART?
Will this chip actually work? If so, how do I remap the UART functions under Arduino (using IDE 2.3.3) and to which pins?
The output protocol is 57600 baud, inverted, half duplex and in fact the (16MHz) ATTiny85 in my original project uses a single pin TX/RX with SoftwareSerial and works flawlessly with both TX and RX on the same physical pin. It even inverts the signal without additional hardware.
Inversion of the signal is not a problem; I already designed an inverter/multiplexer for a hardware UART which I published it here:
https://www.rcgroups.com/forums/showthread.php?4511273-DIY-inverter-for-Pixhawk-to-FrSky%C2%92s-S-PORT-telemetry

So, to complete my project I need either SoftwareSerial to work or to be able to use the hardware UART in the CH32V003J4M6.
Please don't tell me to use another (bigger) version of the CH32V003 chip; the SOP8 ATTiny version works and I also designed an ATMega328P version if a bigger chip is the solution.

maxint-rd, 2024-10-08:
Interesting project! Personally I kind of like to be challenged by the limitations of smaller/cheap processors and the CH32V003 offers some great features at very low cost. Having bought 50 sop8 chips myself, I also think that it would be nice to have SoftwareSerial working. Unfortunately I don't have enough spare time to look into that library anytime soon, but perhaps later this year.

Btw. For one of the chips supported by my TM16xx library I bitbanged single wire fixed timing TX (without crystal), so I believe that should be doable. Having half duplex RX on the same wire probably requires using some more complex programming with a pin-change interrupt implementation. Feel free to look at the code on my github; I think it was rhe TM1652 chip.

For the ATtiny13A I once wrote a pure C/C++ software I2C slave library. That chip ran at 10Mhz and was limited to 100kHz polling based I2C. Interrupts were too slow for those speeds on the T13, but that should be no issue for the faster 48Mhz clock of the CH32V003.

blopa1961, 2024-10-08:
> Edit: the SOP8 does have usable hardware UART. Pin 1 is RX and pin 8 is TX.

In my project PIN1 is OSCI and thus it's connected to the 24MHz crystal.
Which oscilator configuration did you use? Are we still talking Arduino IDE?

maxint-rd, 2024-10-09:
In my test with hardware UART I used the internal oscillator, with normal 48MHz clock-speed. Yes, I used a standard serial example, from within  the Arduino IDE.
I have not yet tried remapping pins for serial.

Have you already tried using the internal oscillator for this project? If you can use the internal osscilator you would free up pins 1 and 3. If not, you only have pins 7 and 8 left for UART after using 5 and 6 for I2C and you woukd indeed need some sort of software serial.

yangliangquan, 2025-04-24:
There is indeed a conflict between the UART and crystal oscillator pins of this package. If you really want to use the external crystal oscillator and the serial port together without conflict, it is recommended to replace it with a slightly larger package, such as the A4M6 or F4U6 in the same series. Their package sizes are also quite small.

closed 2025-04-24 by yangliangquan