openwch/arduino_core_ch32 #159: CH32x033F8P6 How to use multiple serial ports

vuhazu, 2024-10-13:
i redefined serial pin ( PA10, PA11 )
it works, but can't use Serial2 or Serial3

// UART Definitions
#ifndef SERIAL_UART_INSTANCE
  #define SERIAL_UART_INSTANCE  1
#endif
// Default pin used for generic 'Serial' instance
// Mandatory for Firmata
#ifndef PIN_SERIAL_RX
  #define PIN_SERIAL_RX         PA11
#endif
#ifndef PIN_SERIAL_TX
  #define PIN_SERIAL_TX         PA10
#endif

maxint-rd, 2024-10-13:
Interesting! 

In the past days I've been testing my CH32X033F8P6  chips. I made a new variant within the CH32X035 board to improve use of digitalWrite, analogWrite and analogRead. (intend to make that a PR sometime soon).

I tested Serial only on pins PA2/PA3 by setting PIN_SERIAL_RX and PIN_SERIAL_TX near line 113 in the file variant_CH32X033F8P6 .h, deep down the /variants folder. A few lines higher there is a definition for SERIAL_UART_INSTANCE. It is set to 1. I have not dived into that yet, but maybe that definition affects the available serial ports or perhaps only the active port. I've seen that the V203 QT PY sets SERIAL_UART_INSTANCE to 2...

BTW. My PR #145 for interrupt driven Serial1 (with available() peek() and improved read()) was developed for the V003, but other people confirmed it working on several others, including the X035. I haven't tested that functionality on the X033 yet (currently testing EEPROM emulation). However, that PR only affects Serial1.

vuhazu, 2024-10-14:
> Interesting!
>
> In the past days I've been testing my CH32X033F8P6 chips. I made a new variant within the CH32X035 board to improve use of digitalWrite, analogWrite and analogRead. (intend to make [...]

i modified the CH32x035 file to use for CH32x033
can you share the variant file for CH32x033F8P6?

maxint-rd, 2024-10-14:
> i modified the CH32x035 file to use for CH32x033 can you share the variant file for CH32x033F8P6?

Yes I will. Let me do some cleanup and then I'll submit a PR that you can test. I modified multiple files, also to have logical numeric pin numbering and have working PWM and ADC on most pins. (I shall post a link here when ready)

maxint-rd, 2024-10-15:
FYI I've uploaded a first version of my [CH32X033F8P6 variant files](https://github.com/maxint-rd/arduino_core_ch32/tree/support-CH32X033F8P6/variants/CH32X035/CH32X033F8P) in  [this branch](https://github.com/maxint-rd/arduino_core_ch32/tree/support-CH32X033F8P6). It may need some more cleanup here and there...

vuhazu, 2024-10-15:
> FYI I've uploaded a first version of my [CH32X033F8P6 variant files](https://github.com/maxint-rd/arduino_core_ch32/tree/support-CH32X033F8P6/variants/CH32X035/CH32X033F8P) in [this [...]

yes i will try it and will let you know.

maxint-rd, 2024-10-15:
Okay. Next to the variant files I changed [boards.txt](https://github.com/maxint-rd/arduino_core_ch32/blob/support-CH32X033F8P6/boards.txt) and added a [chip specific readme](https://github.com/maxint-rd/arduino_core_ch32/tree/support-CH32X033F8P6/variants/CH32X035/CH32X033F8P). Have fun!

(As soon as I find some more time, I want to test out the serial stuff this issue actually is about...)

danielkucera, 2025-02-14:
I am also trying to use multiple uarts but failing.
To me, following call is suspicious:
https://github.com/openwch/arduino_core_ch32/blob/1a6767a76b533c2bbebefa54a0a4abfd5a591226/cores/arduino/HardwareSerial.cpp#L173
Shouldn't it be using something like `uart_putc(&_serial, &c)` ?
How can it differentiate different ports when the _serial object is not passed in?

maxint-rd, 2025-04-18:
@danielkucera - Tested your PR #180 on CH32X033. Required some changes, but otherwise seems to work. Nice!
(Still struggling to get RX on PA3 working properly)

maxint-rd, 2025-04-19:
Update: got using multiple UART instances working on the CH32X033F8P6. My current code supports either select UART1 or UART2 as single instance, or select both instances to be used simultaneous (at cost of 132/136 bytes more flash/ram than using only one instance). Supported:
 - UART1: PA10=TX1_1, PA11=RX1_1 
 - UART2: PA2=TX2, PA3=RX2

I looked at the SSOP20 pinout and supporting more doesn't seem worth the effort now. 
Not supported (yet):
 - UART3: PC18=TX3_1, PC19=RX3_1 (not useful because these are PC18=SWDIO, PC19=SWCLK)
 - UART4: PA5=TX4_1, PA9=RX4_1 (PB0 for TX4 cannot be output)
 
From what I see the implementation might also be useful for CH32X035, but since I don't have that chip, I will not spend effort on it.

maxint-rd, 2025-04-19:
> it works, but can't use Serial2 or Serial3

@vuhazu: I have Serial2 working along Serial1. Serial3 seems not useful as the pins conflict with SWDIO/SWCLK (See previous comment)

closed 2026-10-31 by medtty