openwch/arduino_core_ch32 #163: How to Use UART with CH32V003F4P6 in Arduino IDE?

Nikitanagar, 2024-11-25:
Hello everyone,
I am new to the CH32V003F4P6 microcontroller and am trying to get started with it using the Arduino IDE. Specifically, I want to use the UART functionality to send and receive data over serial communication.

I have the following questions:

1.  How can I configure UART on the CH32V003F4P6 using the Arduino IDE?
-      I am familiar with using UART on other microcontrollers (like the ATmega328P), but I am unsure of the specific setup for the 
        CH32V003F4P6.
-      Are there any specific libraries or configuration steps that I need to follow for this chip in Arduino?

2. Is there any specific setup needed in the Arduino IDE for the CH32V003F4P6?
-     I have installed the CH32V003 board package in the Arduino IDE, but I am unsure about the settings to use for proper 
      communication.

Any help or examples would be greatly appreciated!
Thank you in advance!

maxint-rd, 2024-11-25:
> How can I configure UART on the CH32V003F4P6 using the Arduino IDE?

You can just use it, same as regular Arduino, but make sure to use the latest code of the master branch, since the released version lacks many improvements. Also look into the PR's to find more improvments. I submitted PR #145 to add interrupt driven serial for better communication and support of Serial.available() and Serial.peek().

> Is there any specific setup needed in the Arduino IDE for the CH32V003F4P6?

Yes. Use the released core, then update with improvements (see above).

Nikitanagar, 2024-11-25:
Thank you for your help. Now, I'm trying to use the I2C protocol on a CH32V003F4P6 microcontroller with the Arduino IDE. I'm testing it with the following I2C scanner code:

```
#include <Wire.h>

void setup() {
  Serial.begin (115200);

  // Wait for serial port to connect (for boards like Leonardo)
  while (!Serial) {}

  Serial.println();
  Serial.println("I2C scanner. Scanning ...");
  
  Wire.begin();
  while(true) {
    byte count = 0;
    for (byte i = 8; i < 120; i++) {
      Wire.beginTransmission (i);
      if (Wire.endTransmission () == 0) {
        Serial.print("Found address: ");
        Serial.print(i, DEC);
        Serial.print(" (0x");
        Serial.print(i, HEX);
        Serial.println(")");
        count++;
        delay(1);
      }
    }
    Serial.println("Done.");
    Serial.print("Found ");
    Serial.print(count, DEC);
    Serial.println(" device(s).");
    delay(1000);
    Serial.println("Scanning again...");
  }
}

```
void loop() {}

**Issue:**
The code runs without errors, but the scanner doesn't detect any I2C devices, and the output repeatedly shows 0 devices found.

Setup:

Microcontroller: CH32V003F4P6
IDE: Arduino IDE (with CH32 Arduino core installed)
Connections:
SCL (I2C clock): PC2
SDA (I2C data): PC1

Tested Devices: [e.g., I2C L3G4200D, I2C 0.91 display]

Steps Taken:
Verified wiring and pull-up resistors.
Confirmed that the I2C device works on other boards like an esp32.
Tried different I2C device addresses and baud rates.

Questions:

Is there any additional configuration required in the CH32V003 Arduino core for I2C to work properly?
Are there known limitations or modifications required in the Wire library for the CH32V003?
Could this be related to the MCU's I2C hardware or pin configuration?
Any help or guidance on debugging this issue would be greatly appreciated. Thank you!

maxint-rd, 2024-11-25:
I2C should work. Most limitations were fixed by my PR #108 (which has been merged).
It looks like you're using the scanner example that I included and that I've tested to work.

My first guess is that you probably need some better pull-ups on the I2C lines. If I remember well, this core doesn't set the internal pull-ups (or perhaps they are insufficient).  So in contrast to using I2C on for instance an Arduino Uno you need to add external pull-ups. I've had fine results using 10K resistors between SDA/SCL and VCC. What values are you using? What level is your VCC?

When developing the I2C PR I've tested with a various devices, such as DS3231 and 24C04.
Sometimes long jumpwires with iffy breadboard connections cause noise on the line. 
I used a logic analyzer to check my readings.You may also want to use one to see what is going on...

Nikitanagar, 2024-11-26:
VCC is 3.3V and I have tried with 4.7K resistor and 10K resistor. Device is still not detecting(L3G4200D is working with other controllers).

maxint-rd, 2024-11-26:
Perhaps you have other I2C devices to test?
Try to use the I2C 0.91 display at 5V. The ones I googled work on both 3v3 and 5v. 
The L3G4200D is only 3v.. Looking at the Sparkfun board it has onboard 10K pull-ups. In that case you don't need additional pull-ups. 

My test was with default I2C speed (100K).

Nikitanagar, 2024-11-26:
Can I use GSM with CH32V003F4P6 ?

maxint-rd, 2024-11-26:
> Can I use GSM with CH32V003F4P6 ?

That question is so board that I can answer yes, sure. Some good duct tape will fix almost anything to your GSM... ;-)

Seriously though: perhaps narrow your question? Is this still related to using UART? Perhaps something for a new issue?

Nikitanagar, 2024-11-26:
yes, it's related to using UART.

maxint-rd, 2024-11-26:
> yes, it's related to using UART.

Then yes; just send the right commands to do whatever you want and keep your code small enough to fit it all in the V003. Good luck with your project!

Nikitanagar, 2024-11-26:
Thank you @maxint-rd  for your help, I was trying to do the interfacing between GSM and CH32V003F4P6, and it's working fine.
Here is the code :
`#include <GSMSimCall.h>

// Use HardwareSerial (USART1) on CH32V003
#define RESET_PIN 4  // Make sure the RESET_PIN is correctly mapped to an available pin

static volatile int num = 0;

// Initialize GSMSimCall with Hardware Serial (USART1)
GSMSimCall call(Serial1, RESET_PIN);  // Serial1 is used for GSM communication

void setup() {
  // Initialize Serial communication for debugging
  Serial.begin(115200);  // For Serial Monitor output
  while (!Serial) {
    ;  // Wait for the serial port to be ready
  }

  // Initialize Serial1 for GSM communication (USART1)
  Serial1.begin(115200);  // Adjust baud rate if necessary for your GSM module
  while (!Serial1) {
    ;  // Wait for GSM module to connect
  }

  // Initialize GSM module
  call.init();

  Serial.print("Set Phone Function... ");
  Serial.println(call.setPhoneFunc(1));  // Set phone function to full functionality
  delay(1000);

  // Check network registration
  sendATCommand("AT+CREG?");
  
  Serial.print("is Module Registered to Network?... ");
  Serial.println(call.isRegistered());
  delay(1000);

  // Check signal quality
  Serial.print("Signal Quality... ");
  Serial.println(call.signalQuality());
  delay(1000);

  // Get operator name from SIM card
  Serial.print("Operator Name... ");
  Serial.println(call.operatorNameFromSim());
  delay(1000);

  // Initialize call function
  Serial.print("Init Call... ");
  Serial.println(call.initCall());
  delay(1000);

  // Make a call
  Serial.print("Call any number... ");
  Serial.println(call.call("+_ _**********"));  // Replace with the target number
}

void sendATCommand(const char* command) {
  Serial1.println(command);  // Send AT command to GSM module via Serial1 (USART1)
  delay(1000);                // Wait for the response
  Serial.println("Sent: ");
  Serial.println(command);    // Print the sent command for debugging
}

void loop() {
  // Handle incoming data from GSM module
  if (Serial1.available()) {
    String buffer = "";
    buffer = Serial1.readString();
    num = num + 1;
    Serial.print(num);
    Serial.print(". ");

    // If there is an incoming call, process it
    if (buffer.indexOf("+CLCC:") != -1) {
      Serial.println(call.readCurrentCall(buffer));  // Handle current call
    } else {
      Serial.println(buffer);  // Print other responses
    }
  }
}
`
I hope someone will find this helpful.

maxint-rd, 2024-11-26:
Excellent! Thanks for the feedback.

I saw the GSM code you posted initially. Seems like a good example that other people might find useful too. Perhaps you want to leave a link here to that example somewhere on GitHub?

flaschenpost, 2025-06-03:
> You can just use it, same as regular Arduino, but make sure to use the latest code of the master branch, since the released version lacks many improvements.

@maxint-rd Could you please give any link how to use the main branch in arduino IDE? I saw you contribution with the clock speed, but I have no idea how to get more recent than the 1.0.4 release.

Tanks!
Marco