openwch/arduino_core_ch32 #165: CH32X035 : use CDC to PC trough USB , both direction working

trimarco232, 2024-12-07:
Hi all ,
Some people have managed to print from the chip to the PC (arduino monitor , for example).
But I did not find any code allowing to send CDC datas from the PC to the chip ?
Is there any definitive reason for this , or am I missing something ? thank you

21km43, 2025-01-14:
I tried to support USB CDC on CH32X035 and created PR in TinyUSB. But it doesn't work properly now by some bugs.

https://github.com/adafruit/Adafruit_TinyUSB_Arduino/pull/486

jobitjoseph, 2025-09-04:
> Hi all , Some people have managed to print from the chip to the PC (arduino monitor , for example). But I did not find any code allowing to send CDC datas from the PC to the chip ? Is there any definitive reason for this , or am I missing something ? thank you

Here is the new CDC library for CH32X035. You can either manually install it or install it through the Arduino library manager. 
https://github.com/jobitjoseph/CH32X035_USBSerial

maxint-rd, 2025-09-04:
Excellent! Thanks for your contribution.
(Related issues: #33, #61)

trimarco232, 2025-09-05:
great thanks jobitjoseph !

(and now the PIOC ... )