openwch/arduino_core_ch32 #166: When will the new version be released?

ziv2013, 2024-12-10:
Hi，

When will the new version be released? I'd like to try the USB function.

Z.t

maxint-rd, 2024-12-15:
Good question.  I'm sure more people would like to know....
All releases are made by @TianpeiLee; and unfortunately he can only do this in his spare time. 

So let's take a pragmatic approach. Do you mean the release of code currently in the master branch or submitted as a PR? If the function you want to try is available in the current master or in a PR, I suggest you not to wait for a new release, but just copy the code over.

If there is no code available yet, it can mean that the function is not supported by this core. Is it supported somewhere else? What processor are you using?

ziv2013, 2024-12-16:
As you know USB is the talent of WCH, I'd like to try TinyUSB function. But I don't know how to use it now. After I downloaded the main branch, and overwrite the code in arduino_core_ch32 directory, I got a lot of error.

maxint-rd, 2024-12-16:
Unfortunately I have not tested any USB functionality myself yet. What MCU are you using? 
From what I read the TinyUSB features where primarily aimed at CH32V203, but other CH32 members are supported too. 

You can read more about that effort in this merged PR: [Add Adafruit qtpy ch32v203 board #136](https://github.com/openwch/arduino_core_ch32/pull/136). Related are merged PR's #137, #127 and #121.

Additional work was done in PR's that haven't been merged yet: [Added LANA TNY board #164](https://github.com/openwch/arduino_core_ch32/pull/164).
Just a few hours ago a new commit was added to [Add functions for USBPD & TinyUSB Update #144](https://github.com/openwch/arduino_core_ch32/pull/144) .

ziv2013, 2024-12-17:
I'm making open source hardware on Ch32V305 now. How can I enabled USB support on this board?

Thanks.

lyusupov, 2024-12-17:
@ziv2013 

FYI,  current state of "master" repo works fine to me on V307 USB in FS-mode. 
HS-mode with proper MHz setting does not enumerate this device.

yangliangquan, 2025-04-28:
I've just started taking over the task of helping to improve this project. After fully understanding it and resolving some of the issues, I'll update a new version as soon as possible.

maxint-rd, 2025-04-28:
@yangliangquan , Thank you for your efforts so far. Good to see things are progressing. I think the CH32 chip family is a very nice addition to the Arduino ecosystem that can help many makers.

I have submitted some PR's in the past that can be useful. Past weeks I improved interrupt driven serial to support dual serial ports on the X033. I intend to do a bit more testing and then I will publish those changes too.

medtty, 2026-10-28:
Tracking the release in #259

closed 2026-10-28 by medtty