openwch/arduino_core_ch32 #168: CH32V307 dev kit

WhoTookMyNicks, 2024-12-20:
Hello, 
I have connected two I2C modules (oled and bme280) to this board and the I2C scanner shows me this:

> I2C Scanner
> Scanning...
> Unknown error at address 0x01
> Unknown error at address 0x02
> Unknown error at address 0x03
> Unknown error at address 0x04
> Unknown error at address 0x05
> Unknown error at address 0x06
> Unknown error at address 0x07
> Unknown error at address 0x08
> Unknown error at address 0x09
> Unknown error at address 0x0A
> Unknown error at address 0x0B
> Unknown error at address 0x0C
> Unknown error at address 0x0D
> Unknown error at address 0x0E
> Unknown error at address 0x0F
> Unknown error at address 0x10
> Unknown error at address 0x11
> Unknown error at address 0x12
> Unknown error at address 0x13
> Unknown error at address 0x14
> Unknown error at address 0x15
> Unknown error at address 0x16
> Unknown error at address 0x17
> Unknown error at address 0x18
> Unknown error at address 0x19
> Unknown error at address 0x1A
> Unknown error at address 0x1B
> Unknown error at address 0x1C
> Unknown error at address 0x1D
> Unknown error at address 0x1E
> Unknown error at address 0x1F
> Unknown error at address 0x20
> Unknown error at address 0x21
> Unknown error at address 0x22
> Unknown error at address 0x23
> Unknown error at address 0x24
> Unknown error at address 0x25
> Unknown error at address 0x26
> Unknown error at address 0x27
> Unknown error at address 0x28
> Unknown error at address 0x29
> Unknown error at address 0x2A
> Unknown error at address 0x2B
> Unknown error at address 0x2C
> Unknown error at address 0x2D
> Unknown error at address 0x2E
> Unknown error at address 0x2F
> Unknown error at address 0x30
> Unknown error at address 0x31
> Unknown error at address 0x32
> Unknown error at address 0x33
> Unknown error at address 0x34
> Unknown error at address 0x35
> Unknown error at address 0x36
> Unknown error at address 0x37
> Unknown error at address 0x38
> Unknown error at address 0x39
> Unknown error at address 0x3A
> Unknown error at address 0x3B
> Unknown error at address 0x3C
> Unknown error at address 0x3D
> Unknown error at address 0x3E
> Unknown error at address 0x3F
> Unknown error at address 0x40
> Unknown error at address 0x41
> Unknown error at address 0x42
> Unknown error at address 0x43
> Unknown error at address 0x44
> Unknown error at address 0x45
> Unknown error at address 0x46
> Unknown error at address 0x47
> Unknown error at address 0x48
> Unknown error at address 0x49
> Unknown error at address 0x4A
> Unknown error at address 0x4B
> Unknown error at address 0x4C
> Unknown error at address 0x4D
> Unknown error at address 0x4E
> Unknown error at address 0x4F
> Unknown error at address 0x50
> Unknown error at address 0x51
> Unknown error at address 0x52
> Unknown error at address 0x53
> Unknown error at address 0x54
> Unknown error at address 0x55
> Unknown error at address 0x56
> Unknown error at address 0x57
> Unknown error at address 0x58
> Unknown error at address 0x59
> Unknown error at address 0x5A
> Unknown error at address 0x5B
> Unknown error at address 0x5C
> Unknown error at address 0x5D
> Unknown error at address 0x5E
> Unknown error at address 0x5F
> Unknown error at address 0x60
> Unknown error at address 0x61
> Unknown error at address 0x62
> Unknown error at address 0x63
> Unknown error at address 0x64
> Unknown error at address 0x65
> Unknown error at address 0x66
> Unknown error at address 0x67
> Unknown error at address 0x68
> Unknown error at address 0x69
> Unknown error at address 0x6A
> Unknown error at address 0x6B
> Unknown error at address 0x6C
> Unknown error at address 0x6D
> Unknown error at address 0x6E
> Unknown error at address 0x6F
> Unknown error at address 0x70
> Unknown error at address 0x71
> Unknown error at address 0x72
> Unknown error at address 0x73
> Unknown error at address 0x74
> Unknown error at address 0x75
> Unknown error at address 0x76
> Unknown error at address 0x77
> Unknown error at address 0x78
> Unknown error at address 0x79
> Unknown error at address 0x7A
> Unknown error at address 0x7B
> Unknown error at address 0x7C
> Unknown error at address 0x7D
> Unknown error at address 0x7E
> No I2C devices found
> 

The I2C scanner I started with Wire.begin(PB10, PB11); //scl, sda, as I connected the I2C devices here.

maxint-rd, 2024-12-20:
Do you have release 1.0.4 updated with the latest code from the master branch?
For the CH32V003 I implemented merged PR #108 to support I2C scanning (and I2C slave).
Unfortunately I don't have the CH32V307, so I couldn't test this on that CPU.

With other CH32 chips I had also to ensure that the I2C lines had pull-ups, and that I2C was enabled.
In other experiments I encountered issues when using fast speed with iffy breadboard connections and long jumper wires. In some cases the power supplied gave issues and I had to add a capacitor or use 5V instead of 3v3.

WhoTookMyNicks, 2024-12-20:
It is 1.0.4
Interestingly, in the package examples, there is no I2C scanner.
I tried your example from here but only added Wire.begin(PB10, PB11);
Now I get this:

> Found 0 device(s).
> Scanning again...
> Done.
> 

Wires are some 15cm, 5V. It's still the same.

maxint-rd, 2024-12-20:
> Interestingly, in the package examples, there is no I2C scanner.

To me that indicates that you only installed the released core, not the latest updates that can be found in the master branch. For the I2C update you need to copy the latest code of the master branch over the installed 1.0.4 code.
(BTW. In issue #124 someone initially had problems compiling for the v30x, but that was resolved in PR #126 and tested successfully. I'm not sure if @lyusupov also tested the scanning or what precise MCU he used).

In case you still have issues after updating the code:
- Do you have a logical analyzer that you can use to look at the I2C signals? 
To support scanning timeouts were reduced and on more occasions stops are generated. Using the analyzer you can have a deeper look.

- On CH32V307 PB10, PB11 are the I2C2 pins. Have you tried the standard I2C pins? (I2C1 SDA=PB7, SCL=PB6). I mean can you try Wire.begin() without parameters?
The V003 only has one I2C port, so I have not tested scanning on second I2C channels.

- My tests were with default I2C speed (100K). You could try slower speeds.

lyusupov, 2024-12-20:
@WhoTookMyNicks 

I would not recommend to use Release 1.0.4 - try to apply "master" branch instead.
There are numerous Core fixes and updates since December 2023 (1.0.4 release date)

These are V307 Core features been qualified as doing quite good to me.
https://github.com/lyusupov/SoftRF/releases/tag/1.6

[image]

Someone may consider to pick some tips from my HAL source code:
1) https://github.com/lyusupov/SoftRF/blob/master/software/firmware/source/SoftRF/src/platform/CH32.cpp
2) https://github.com/lyusupov/SoftRF/blob/master/software/firmware/source/SoftRF/src/platform/CH32.h

WhoTookMyNicks, 2025-02-15:
Sorry for late reply.

How to apply a master branch? I can not install it in the Arduino IDE using the JSON package URL? How to replace it with the one I have (1.0.4)?

maxint-rd, 2025-02-15:
> How to apply a master branch? I can not install it in the Arduino IDE using the JSON package URL? How to replace it with the one I have (1.0.4)?

First install the 1.0.4 release, then copy all files from the master branch to their specific location. Finally determine which unmerged PR's you may need and copy those files too.

medtty, 2026-10-28:
Fixed on main (Wire updates in #108 / #126 ) and will be in the next release. Closing; please reopen if it still happens on the new release.

closed 2026-10-28 by medtty

medtty, 2026-10-28:
Thanks @DevinJM3  and thank you for setting up the ch32-community fork and keeping things moving while this repo was quiet. I’m with WCH and now maintaining this repository, going through the open issues and PRs, testing on hardware, and preparing a new release with a new board support. I’ve looked at your fork and the CI you’ve added; I’d like to bring that work back here rather than have the two drift apart, and will follow up once the approach is agreed internally.