openwch/arduino_core_ch32 #170: GPIO   - digitalWrite(PC11,0x00);

gitCittius, 2025-01-03:
Good Afternoon! I have a CH32V307 EVT development board and I cannot configure the pins available on the board. Only those pins that are not in the Arduino format work.

maxint-rd, 2025-01-03:
Unfortunately I don't have the V307. For the CH32 chips I do have I did some testing to see which Arduino pin number matched which CH32 pin. It was just a simple for loop that did a digitalWrite to each pin number.  

This was the result for the CH32V003F4P6 TSSOP20 
```
        +-----v-----+
  14 1 -+PD4     PD3+- 20  13
  TX 2 -+PD5     PD2+- 19  12
  RX 3 -+PD6     PD1+- 18  SWIO
 RST 4 -+PD7     PC7+- 17  9
  0  5 -+PA1     PC6+- 16  8
  1  6 -+PA2     PC5+- 15  7
 GND 7 -+VSS     PC4+- 14  6
 10? 8 -+PD0     PC3+- 13  5
 VCC 9 -+VDD     PC2+- 12  4/SCL
  2 10 -+PC0     PC1+- 11  3/SDA
        +-----------+
```

Of course the pinout is different for SOP8 and SOP16 packages. 

For the CH32X033 I went through the same exercise, but first had to make a [board definition](https://github.com/maxint-rd/arduino_core_ch32/tree/support-CH32X033F8P6/variants/CH32X035/CH32X033F8P) that worked for that chip. For the CH32X033 TSSOP20 I made this fine piece of ASCII art:
```
                  +------v------+
    MISO/A6 D6~ 1-+PA6       PA5+-20  D5 SCK/A5
MOSI/A8/TX4 D7  2-+PA7=PB0   PA4+-19  D4~ CS/A4
     A9/RX4 D8  3-+PB1      PC19+-18  D17 SWCLK
       /RST D9  4-+PB7       PA3+-17  D3~ RX2/A3*
     USBDM D10  5-+PC16=PC11 PA2+-16  D2~ TX2/A2
     USBDP D11  6-+PC17=PC10 PA1+-15  D1~ A1
           GND  7-+VSS       PA0+-14  D0~ A0
     SWDIO D16  8-+PC18      PC3+-13  D15~ A13
           VCC  9-+VDD      PA10+-12  D14/SCL
           D12 10-+PA9      PA11+-11  D13/SDA
                  +-------------+
*A3 and I2C don't work on CH32X033F8P6 0-series (lot number with the penultimate bit 5 being 0).
```

Since that board definition didn't exist yet, I was free to choose which Arduino pin numbers to map to which CH32 pins. Starting with pin number 0 for PA0 seemed most logical and then I followed port A and went counter clockwise for the others.

lyusupov, 2025-01-04:
@gitCittius PC11 is defined in this PR https://github.com/openwch/arduino_core_ch32/pull/161
read this commit https://github.com/openwch/arduino_core_ch32/pull/161/commits/2aec0416e852d2f30672034303db9c453b0c4045 for more details

WhoTookMyNicks, 2025-02-15:
> Good Afternoon! I have a CH32V307 EVT development board and I cannot configure the pins available on the board. Only those pins that are not in the Arduino format work.

https://forum.arduino.cc/t/ch32v307vct6-development-board-design/1333104

medtty, 2026-10-28:
Fixed by #161 , which adds PC11 and the other remaining pins (D22–D32) to the CH32V307VCT6 variant

closed 2026-10-28 by medtty