openwch/arduino_core_ch32 #173: libusb problems on apple silicon

LeonardoCT, 2025-01-05:
Hi,
After installing libusb using brew this error message still happears:

`Sketch uses 4392 bytes (26%) of program storage space. Maximum is 16384 bytes.
Global variables use 388 bytes (18%) of dynamic memory, leaving 1660 bytes for local variables. Maximum is 2048 bytes.
dyld[5289]: Library not loaded: /usr/local/opt/libusb/lib/libusb-1.0.0.dylib
  Referenced from: <C79B1DDB-9DAE-30AE-9474-2E146265EC5A> /Users/leonardoferlito/Library/Arduino15/packages/WCH/tools/openocd/1.0.0/bin/openocd
  Reason: tried: '/usr/local/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/usr/local/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), '/usr/local/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), '/usr/local/lib/libusb-1.0.0.dylib' (no such file), '/usr/lib/libusb-1.0.0.dylib' (no such file, not in dyld cache)
Failed uploading: uploading error: signal: abort trap`

This is due to a broken link, with the use of a symlink you can force the system to use the arm version (installed by brew) of libusb then it complains about it being the wrong architecture

LeonardoCT, 2025-01-05:
This is what happens when you "force it" by using the symlink:

`dyld[7861]: Library not loaded: /usr/local/opt/libusb/lib/libusb-1.0.0.dylib
  Referenced from: <C79B1DDB-9DAE-30AE-9474-2E146265EC5A> /Users/leonardoferlito/Library/Arduino15/packages/WCH/tools/openocd/1.0.0/bin/openocd
  Reason: tried: '/usr/local/opt/libusb/lib/libusb-1.0.0.dylib' (mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64')), '/System/Volumes/Preboot/Cryptexes/OS/usr/local/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), '/usr/local/opt/libusb/lib/libusb-1.0.0.dylib' (mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64')), '/usr/local/lib/libusb-1.0.0.dylib' (no such file), '/usr/lib/libusb-1.0.0.dylib' (no such file, not in dyld cache)
Failed uploading: uploading error: signal: abort trap`

`**(mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64'))**`

i don't have the knowledge to fix this, may be a openocd needs to be ported to arm64?

LeonardoCT, 2025-01-05:
Update:

I did try installing the apple silicon native version of openocd (using homebrew) and poiting the bundled-executable to the arm64 executable, but unfortunately the bundled version is a custom one and therefore the version found on brew isin't compatibile:

`Open On-Chip Debugger 0.12.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Error: The specified debug interface was not found (wlinke)
The following debug adapters are available:
1: dummy
2: ftdi
...
...
Failed uploading: uploading error: exit status 1`

Looks like that this issue could be fixed by recompiling the original source code and get an apple silicon native version should the source code be avaible

DeqingSun, 2025-01-06:
As a person that don't really like brew. This is my folder of ```/usr/local/Cellar/libusb``` on a 2014 Intel I7 mac and I copied the files to a M2 computer. It works just fine.

[libusb.zip](https://github.com/user-attachments/files/18313254/libusb.zip)

LeonardoCT, 2025-01-06:
Thank you! i had to jump some extra steps with apple quarantine stuff, but now it works! the file you provided is the x86 version of the libusb library and the fact that even if successful a `Failed uploading: uploading error: exit status 1` will appear, and therefore i still consider this solution a workaround and not a proper fix, on my spare time i will try to recompile openocd_wch for apple silicon so that this issue may be fixed, until then, thank you a lot!

DeqingSun, 2025-01-06:
You error sounds like

https://github.com/openwch/arduino_core_ch32/issues/49

LeonardoCT, 2025-01-15:
> You error sounds like
> 
> #49

I can confirm that downloading the MSR toolchain for mac from the mentioned post will fix all the issues with libusb and the "exit status 1 error", all that is needed is to download the MRS Toolchain and browse the arm64 folder and copy the "openocd" executable to the right folder (where arduino stores all the boards tools and utilities), for the next release of the arduino core this precopiled arm64 should be included in the default installation

closed 2025-01-15 by LeonardoCT

dchicchon, 2025-04-25:
Hey @LeonardoCT I'm getting the same issue on my end with the libusb pointing to the missing `/usr/local/opt` directory on my  M1 mac. By any chance did you end up resolving the issue with the symlink?

dchicchon, 2025-04-25:
On my end after downloading the provided libusb file from @DeqingSun , I end up getting this result in my uploader

```
Sketch uses 27720 bytes (12%) of program storage space. Maximum is 229376 bytes.
Global variables use 4340 bytes (42%) of dynamic memory, leaving 5900 bytes for local variables. Maximum is 10240 bytes.
Open On-Chip Debugger 0.11.0+dev-00013-gcb033e0 (2023-12-18-14:57)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'sdi'
Warn : Transport "sdi" was already selected
Ready for Remote Connections
Error: open failed

Failed uploading: uploading error: exit status 1
```