openwch/arduino_core_ch32 #179: Flash Memory Optimization for CH32V003

thevien2507, 2025-02-11:
I'm working with the CH32V003 microcontroller and noticed that peripheral libraries (Wire for I2C, SPI, and UART) consume a significant amount of flash memory.  

For example, just using the Wire library for I2C and Serial takes up about **73%** of the available flash memory (**12,018 bytes out of 16,384 bytes**).  

Here is a minimal example of my I2C code:  

```cpp
#include <Wire.h>

void setup() {
    Wire.begin(8);  // Initialize as I2C slave with address 8
    Wire.onReceive(receiveEvent);
    Serial.begin(9600);
}

void loop() {
    // Do nothing, waiting for data from master
}

void receiveEvent(int howMany) {
    Serial.print("Received: ");
    while (Wire.available()) {
        char c = Wire.read();
        Serial.print(c);
    }
    Serial.println();
}
```

### **Flash & RAM Usage After Compilation:**  
```
Sketch uses 12,108 bytes (73%) of program storage space. Maximum is 16,384 bytes.  
Global variables use 680 bytes (33%) of dynamic memory, leaving 1,368 bytes for local variables. Maximum is 2,048 bytes.  
```

I also noticed that SPI and UART libraries take up a lot of flash memory as well, making it difficult to fit multiple peripherals into the available 16 KB flash.  

### **Comparison with MounRiver**  
When compiling a similar I2C example using MounRiver Studio, the flash usage is significantly lower:  

```
   text    data    bss    dec    hex    filename  
   4196     64    260   4520   11A8    I2C_7bit_Mode.elf  
```

This means that MounRiver only uses **~4.2 KB flash**, while the Arduino Core version consumes **~12 KB**, nearly **3 times more**.  

### **Questions:** 
- Are there any compiler flags or configuration options to disable unnecessary features and optimize resource usage?  
- Is there any way to optimize flash usage while keeping I2C, SPI, and UART functionality?  

Any suggestions for reducing flash usage would be greatly appreciated! 🚀

maxint-rd, 2025-02-11:
Hello @thevien2507 ; you've encountered the same issue that I have. Insuffiecient flash space is the largest challenge I've faces in my CH32V003 projects. Since your example uses I2C slave, I assume you are using a recent IDE with latest core (i.e master branch plus all relevant PR's).

I've not found a definite solution, but this is what I do to combat this:
1. Switch off all unused features; UART, SPI, I2C, hardware timer, etc. The official way at the moment is to change the defines that are somewhere in a header file under system (don't know exactly by heart; it's mentioned in other related issues). For I2C-slave there is a separate define in the library to switch that featue off.
(I've modified my header file and board file to make changing options easier using the IDE menu, but don't have these changes working for all CH32 chips, so I won't post it as a PR.)
2. Don't use Serial if you can avoid it. Use digitalWrite to blink a led and a logical analyzer to monitor I2C traffic. If you use serial, use short strings and never use printf. 
3. Use the realtime debugger to debug specific code sections while other parts are switched off using ifdefs.
4. Use the CH32X033 if you need Serial for debugging. Its TSSOP20 package pinout has the closest match to the V003.The X033 has 64kB flash allowing for dev code that can be stripped out later using ifdefs.
5. Use an old Arduino IDE (1.8.19?) that has functioning -LTO option to produce smaller flash usage. I've not found the cause yet why Link Time Optimization doesn't work on the 2.3.2 IDE with latest core, while it does work on the old IDE.

svofski, 2026-08-19:
I also stumbled upon steep increase of flash usage after adding Wire.h to a very basic CH32V003 sketch. I traced it back to delayMicroseconds() which uses uint64_t in general case, which pulls in humongous __udivdi3. Here's my version of wiring_time.h optimized for CH32V00x (I'm using it with CH32V003F4P6). Full disclosure: I didn't verify correctness except that I got my sketch to fit in flash and I2C slave to respond on the bus.
```
#if defined(CH32V10x)
... (special CH32V10x version)...
#elif defined(CH32V00x)
static inline void delayMicroseconds(uint32_t) __attribute__((always_inline, unused));
static inline void delayMicroseconds(uint32_t us)
{
  __IO uint32_t currentTicks = SysTick->CNT;
  /* Number of ticks per millisecond: this still calls _udivsi3, would be nice if we could cache it once maybe in SystemCoreClockUpdate ? */
  uint32_t tickPerMs = SystemCoreClock / 8000000ULL; // would be great not to have to compute this value every time
  uint32_t nbTicks = (us - ((us > 0) ? 1 : 0)) * tickPerMs;
  /* Number of elapsed ticks */
  uint32_t elapsedTicks = 0;
  __IO uint32_t oldTicks = currentTicks;
  do {
    currentTicks = SysTick->CNT;

    //increment
    elapsedTicks += (oldTicks <= currentTicks) ? currentTicks - oldTicks :
                     tickPerMs - oldTicks + currentTicks;

    oldTicks = currentTicks;
  } while (nbTicks > elapsedTicks);  
}
#else
...(default non-specific version)...
#endif
```
This still uses _udivsi3, but it's used in multiple other places so not a problem. Wasting time on _udivsi3 every time you do a tiny delay seems annoying though. I see two possibilities:
  - there is a nice Delay_Us(uint32_t) in `system/CH32V00x/SRC/Debug/debug.c`, but it relies on a cached ticks per us and ticks per ms values and needs a call to `Delay_Init()` so not a drop-in replacement for delayMicroseconds().  But `SystemCoreClockUpdate()` could call `Delay_Init()`, that would be nice. Only 3 bytes lost to cached precomputed values.

  - another alternative is to just use F_CPU define in place of SystemCoreClock, then the whole expression for tickPerMs will evaluate at compile time like it used to in good old avr-gcc. But I'm not sure about all possible clock options on CH32V003, probably a risky choice for generic library code.

maxint-rd, 2026-08-20:
Hello @svofski , as mentioned in my changes for PR #200, I noticed the same usage of costly 64-bit code. The easiest fix for me was to use 32-bit variables where applicable. In my opinion minimizing flash is kind of essential for all of the smallest processors. For that reason I specified both `CH32VM00X` and `CH32V00x`. Another big thing for me was to get LTO working (see PR #215). Additional smaller flash savings are mentioned in [this comment](https://github.com/openwch/arduino_core_ch32/issues/204#issuecomment-3015173859) and [this comment](https://github.com/openwch/arduino_core_ch32/issues/204#issuecomment-3140175460).

I didn't compare your changes to mine, my main change was to simply change the data size from 64-bit to 32-bit. For reference I list my changes below.

------------

For `getCurrentMicros()` I made [this change](https://github.com/maxint-rd/arduino_core_ch32/commit/c06c41f58f6d8ce7ba211d2ee506250ef71a3a41) in clock.c:

```
#if defined(CH32VM00X) || defined(CH32V00x)
  // MMOLE  250708: a uint64_t division causes need for __udivdi3, which is quite large (more than 1KB)
  //      SysTick->CMP and CNT are 32-bit values, let's try to avoid 64 bit divisions to minimize Flash usage
  //      (This might cause smaller roll-over, but who cares...)
  // NOTE: 64 bit divisions are also used by delayMicroseconds() in wiring_time.h
  uint32_t m0 = GetTick();
  __IO uint32_t u0 = SysTick->CNT;
  uint32_t m1 = GetTick();
  __IO uint32_t u1 = SysTick->CNT;   //may be a interruption
   uint32_t tms = SysTick->CMP + 1;

  if (m1 != m0) {
    //return (m1 * 1000 + ((tms - u1) * 1000) / tms);
    return m1 * 1000 + u1 * 1000 / tms;  // fix issue #65
  } else {
    //return (m0 * 1000 + ((tms - u0) * 1000) / tms);
    return m0 * 1000 + u0 * 1000 / tms;  // fix issue #65
  }
#else
```

A [similar change](https://github.com/maxint-rd/arduino_core_ch32/commit/2d8d867c8041fe2591ebbfe5a987e968891ab4f7) was made for `delayMicroseconds()` in wiring_time.h:
```
#if defined(CH32VM00X) || defined(CH32V00x)
  // MMOLE 250708: a uint64_t division causes need for __udivdi3, which is quite large (more than 1KB)
  //      SysTick->CMP and CNT are 32-bit values, let's try to avoid 64 bit divisions to minimize Flash usage
  //      (This might cause smaller roll-over, but who cares...)
  // NOTE: 64 bit divisions are also used by micros() in clock.c
  __IO uint32_t currentTicks = SysTick->CNT;
  /* Number of ticks per millisecond */
  uint32_t tickPerMs = SysTick->CMP + 1;
  /* Number of ticks to count */
  uint32_t nbTicks = ((us - ((us > 0) ? 1 : 0)) * tickPerMs) / 1000;
  /* Number of elapsed ticks */
  uint32_t elapsedTicks = 0;
  __IO uint32_t oldTicks = currentTicks;
  do {
    currentTicks = SysTick->CNT;
    // elapsedTicks += (oldTicks < currentTicks) ? tickPerMs + oldTicks - currentTicks :
    //                 oldTicks - currentTicks;
    
    //increment
    elapsedTicks += (oldTicks <= currentTicks) ? currentTicks - oldTicks :
                     tickPerMs - oldTicks + currentTicks;

    oldTicks = currentTicks;
  } while (nbTicks > elapsedTicks);
```

closed 2026-11-03 by medtty