openwch/arduino_core_ch32 #182: arduino upload fail

652436962, 2025-02-20:
[image]

maxint-rd, 2025-02-20:
What platform are you using? In issue #44 some Windows users had problems with that file. Your path seems to indicate you're using linux (or mac?). For Linux there seems to be some pre-install script dealing with that file. 

At least you've managed to compile something. Maybe you can now try the LinkE utility? Don't know if that even exists for Linux
Unfortunately I only have experience using Windows. Perhaps someone else can be of assistance here?

652436962, 2025-02-21:
> 您使用的是什么平台？在问题[#44](https://github.com/openwch/arduino_core_ch32/issues/44)中，一些 Windows 用户在使用该文件时遇到了问题。您的路径似乎表明您正在使用 Linux（或 Mac？）。对于 Linux，似乎有一些预安装脚本可以处理该文件。
> 
> 至少你已经成功编译了一些东西。也许你现在可以尝试 LinkE 实用程序？不知道 Linux 上是否存在这个 实用程序。不幸的是，我只使用过 Windows。也许其他人可以在这里提供帮助？

linux

652436962, 2025-02-24:
官方回复我了

您好，linux上不通过ide下载hex文件有两种途径。

 

1．使用ISP。

    在官网https://www.wch.cn/downloads/WCHISPTool_CMD_ZIP.html页面下载ISP命令行工具，根据其中说明进行下载。

2.使用两线通讯。

在http://www.mounriver.com/download页面，下载linux工具链与调试器MRS_Toolchain_Linux_x64_V1.92.1.tar.xz，使用其中的OpenOCD进行下载，具体方式可参考README文件。

 

感谢您的反馈，祝您工作顺利。

 

MounRiver Studio

2025.2.24

closed 2025-02-24 by 652436962

652436962, 2025-02-24:
[image]

maxint-rd, 2025-02-24:
Translation for other readers:

> The official replied to me
>
> Hello, there are two ways to download hex files on Linux without ide.
> 
> 1. Use ISP.
> 
> Download the ISP command line tool on the official website https://www.wch.cn/downloads/WCHISPTool_CMD_ZIP.html and download it according to the instructions.
> 
> 2. Use two-wire communication.
> 
> Download the Linux tool chain and debugger MRS_Toolchain_Linux_x64_V1.92.1.tar.xz on the http://www.mounriver.com/download page, and use OpenOCD to download. For specific methods, please refer to the README file.
> 
> Thank you for your feedback and wish you a smooth work.
> 
> MounRiver Studio 2025.2.24