openwch/arduino_core_ch32 #183: ch32v003 Analogwrite not working PD0,PD1,PA2

taisirou, 2025-03-03:
AnalogWrite on PD0, PD1, and PA2 pins does not work.
It looks like a negative pin（ch n pin), but am I using it incorrectly?
I don't know if it's not moving.

taisirou, 2025-03-03:
I think this is an addition to known information.

https://github.com/openwch/arduino_core_ch32/issues/122#issuecomment-2317396484
>PWM SOP16 tested working: PA1/PC0/PC3/PC4 -
>PWM not working: PC1/PC2/PC6/PC7 (only on/off); PD7=/RST; PA2: always off

PWM not working: PD0, PD1: always off

jason3w, 2025-03-15:
I'm unable to use analogWrite on PC5, PC6, PC7. digitalWrite works fine on those pins, but not analogWrite. 
If i understand correctly they should work, right? Thanks!

maxint-rd, 2025-03-15:
> I'm unable to use analogWrite on PC5, PC6, PC7. digitalWrite works fine on those pins, but not analogWrite. If i understand correctly they should work, right? Thanks!

No, they don't. See quote above. (Didn't have time to do another test)

jason3w, 2025-03-15:
Hi Maxint. Thanks for the quick reply. 
I've dug deeper and it looks like tim2 is on PD3, PD4, PD7. I've tried that too, but no luck still. 
Is PWM (analogWrite) not available at all within this platform?

maxint-rd, 2025-03-16:
Yes, PWM is available. as I mentioned in the other post I've successfully used it. 
In one project I used it to regulate the speed of a fan.  I've tested these to work on the SOP16 package: PA1/PC0/PC3/PC4

Note however that in order to use PWM you need to:
 - use the latest release, update it with the latest code of the master branch
 - enable the timer module
 - (optional) review all PR's that have not been merged yet to see what else you may need

I suggest you first try to use one of these ports: PA1/PC0/PC3/PC4.
Good luck!

troudel, 2025-06-12:
Hello community, if someone wants to make chip pins 1, 3, 7 working for analogWrite on CH32V003J4M6 (SOP8), I've been using below config. As chip pin1 is mapped to PD6 and PA1, you cannot use T1CH2 with PA1 as that needs to be used with chip pin 3 mapped to PA2. The solution is using PD6 with T2CH3, below:

TIM_OC3Init( TIM2, &TIM_OCInitStructure );       // on PD6
TIM_OC2Init( TIM1, &TIM_OCInitStructure );       // on PA2
TIM_OC4Init( TIM1, &TIM_OCInitStructure );       // on PC4

Check the datasheet, you need to GPIO_PinRemapConfig(GPIO_FullRemap_TIM2, ENABLE); and don't forget to TIM_OCInitStructure.TIM_OCNPolarity = TIM_OCNPolarity_Low; as PA2 is on the N part of T1CH2N. All these need to be enabled:

dfleck, 2025-07-22:
Doh! This board does have an external crystal. That explains PA1 and PA2.

Also - I missed it but there is an invert flag defined in `PinNamesTypes.h` it is `CH_PIN_INV_BIT` so ignore
`#define	TIMER_INVERTED (CH_PIN_INV_MASK << CH_PIN_INV_SHIFT)`
and change PD0, PA2, and PD1 in `PinMap_TIM[]` to:
```
  {PD_0,       TIM1, CH_PIN_DATA_EXT(CH_MODE_OUTPUT_50MHz | CH_PIN_INV_BIT, CH_CNF_OUTPUT_AFPP, NOPULL, AFIO_Remap_TIM1_DISABLE, 1)}, // TIM1_CH1N
  {PA_2,       TIM1, CH_PIN_DATA_EXT(CH_MODE_OUTPUT_50MHz | CH_PIN_INV_BIT, CH_CNF_OUTPUT_AFPP, NOPULL, AFIO_Remap_TIM1_DISABLE, 2)}, // TIM1_CH2N
  {PD_1,       TIM1, CH_PIN_DATA_EXT(CH_MODE_OUTPUT_50MHz | CH_PIN_INV_BIT, CH_CNF_OUTPUT_AFPP, NOPULL, AFIO_Remap_TIM1_DISABLE, 3)}, // TIM1_CH3N

```
Adding these to `PinMap_TIM[]` will get the rest of your pins working:
```
  {PC_1,       TIM2, CH_PIN_DATA_EXT(CH_MODE_OUTPUT_50MHz, CH_CNF_OUTPUT_AFPP, NOPULL, AFIO_FullRemap_TIM2_ENABLE, 1)}, // TIM2_CH1
  {PC_2,       TIM2, CH_PIN_DATA_EXT(CH_MODE_OUTPUT_50MHz, CH_CNF_OUTPUT_AFPP, NOPULL, AFIO_PartialRemap1_TIM2_ENABLE, 2)}, // TIM2_CH2
  {PC_7,       TIM2, CH_PIN_DATA_EXT(CH_MODE_OUTPUT_50MHz, CH_CNF_OUTPUT_AFPP, NOPULL, AFIO_FullRemap_TIM2_ENABLE, 2)}, // TIM2_CH2
  {PD_6,       TIM2, CH_PIN_DATA_EXT(CH_MODE_OUTPUT_50MHz, CH_CNF_OUTPUT_AFPP, NOPULL, AFIO_FullRemap_TIM2_ENABLE, 3)}, // TIM2_CH3
  {PD_5,       TIM2, CH_PIN_DATA_EXT(CH_MODE_OUTPUT_50MHz, CH_CNF_OUTPUT_AFPP, NOPULL, AFIO_FullRemap_TIM2_ENABLE, 4)}, // TIM2_CH4
  {PC_5,       TIM1, CH_PIN_DATA_EXT(CH_MODE_OUTPUT_50MHz, CH_CNF_OUTPUT_AFPP, NOPULL, AFIO_FullRemap_TIM1_ENABLE, 3)}, // TIM1_CH3
  {PC_6,       TIM1, CH_PIN_DATA_EXT(CH_MODE_OUTPUT_50MHz | CH_PIN_INV_BIT, CH_CNF_OUTPUT_AFPP, NOPULL, AFIO_FullRemap_TIM1_ENABLE, 3)}, // TIM1_CH3
```

dfleck, 2025-07-22:
I haven't tested multiple simultaneous outputs. My understanding of the documentation is that remapping one pin remaps them all per the tables you mention (plus the bit definitions on page 61 seem to confirm this).

Maybe the most user friendly solution would be to have the four groups of pins in separate defines within the `PinMap_TIM[]` structure (in very terse pseudocode form)(the last three being the TimerN pins):
```
#ifdef Timer1_Alternate_Map_1
PC6, PC7, PC0, PD3, PC3, PC4, PD1
#else #ifdef Timer1_Alternate_Map_2
PD2, PA1, PC3, PC4, PD0, PA2, PD1
#else #ifdef Timer1_Alternate_Map_3 // Full remap
PC4, PC7, PC5, PD4, PC3, PD2, PC6
#else //default map
PD2, PA1, PC3, PC4, PD0, PA2, PD1
```
and similar for Timer2. This will keep people for inadvertently intermixing pins from different mappings.

I'm not sure how to keep people from using both the regular and N output at the same time (you can't analogwrite to T1PD2 and T1PD0 at the same time because they use the same timer channel).

dfleck, 2025-07-24:
I just put together a 003J4 test fixture and can verify that with the above changes digitalWrite and analogWrite are working on pins PD6, PA1, PA2, PC1, PC2 and PC4. I did not test PD1, PD4 or PD5 since they are on the SWIO pin. I'm now starting to test simultaneous PWMs.

maxint-rd, 2025-07-25:
If you're doing more development on the J4M6 (SOP8), it might be worthwhile to make a J4 specific board variant that has all the right pin remappings and peripherals specific to that package. It's not very complex to do, but it does take some time to test and fix things.

I did something similar in PR #200, for the V006, where the original board definition was only available for the CH32V006K8U6 (QFN32). Since I bought a bunch of the CH32V006F8P6 (TSSOP20) and also the V002, I wanted to have board variants that closely match the V003 TSSOP20 package. (It works for me, so I though it might be handy for others too...)

dfleck, 2025-07-25:
I have a board definition I've started - but as you say there's a lot of testing that needs to be done before I would want to make an actual pull request. I have tested digitalWrite and analogWrite but I would still need to test analogRead, I2C, UART, and the many functions on the SWIO pin. Plus it's been several years since the last time I actually did any pull requests so I'd have to figure that out again.
If I keep using the chip I'm sure I'll end up using analogRead and I2C, but right now I'm anxiously awaiting for you to merge #204 because the Arduino core is HUGE and causing me memory issues.

I'd be glad to share what I've done so far if anyone wants to complete it.

closed 2026-11-11 by medtty