openwch/arduino_core_ch32 #184: 有支持低功耗蓝牙系列的计划吗

RegularSource, 2025-03-14:


maxint-rd, 2025-03-14:
Translation for other readers:
> Are there plans to support Bluetooth Low Energy series?

DeqingSun, 2025-03-19:
Not official but I have a WIP support for CH573.

https://github.com/DeqingSun/arduino_core_ch32/blob/ee226e1c48288d2f7a41c0c35ad5a9c525595fb3/libraries/CH573BlePeripheral/example/example_ch573_ble_test/example_ch573_ble_test.ino

At this moment, blePeripheral, BLEService, Characteristic with BLERead, BLEWrite, BLENotify works. Serial pinMode, digitalWrite, works.

The upload toolchain is not ready and the whole package is not packed properly, but I'll do it before Fall.

DeqingSun, 2025-04-22:
Ready to use package is on 
https://github.com/DeqingSun/arduino_core_ch32/releases/download/arduinoPack/package_ch32v_index_add_ch573.json

There is BLE example (included in Arduino package) on https://github.com/DeqingSun/arduino_core_ch32/blob/add_ch573/libraries/CH573BlePeripheral/example/example_ch573_ble_test/example_ch573_ble_test.ino

The library is a port of https://github.com/sandeepmistry/arduino-BLEPeripheral , most basic features are ported and functional.

You can even use SerialUSB and program the chip like Arduino Leonardo. The chip can automatically jump to bootloader. The feature does not work perfectly on Windows because the USB on Windows get ready later than WCHISP access. I'll take a look.