openwch/arduino_core_ch32 #186: NO found ws2812 library and example，Pls add

wukongxuetang, 2025-04-01:
Searched and found an issue with tile 'add ws2812 example'(https://github.com/openwch/arduino_core_ch32/issues/11),Closed and completed

But I couldn't find any of them，PLS add one，Thanks

lyusupov, 2025-04-13:
WS1812, also known as NeoPixels, can be used by CH32 with assistance of Adafruit this library
https://github.com/adafruit/Adafruit_NeoPixel/releases/tag/1.12.3

This library was tested by me with CH32V307.and known to work nicely

You may consider to start with very basic example there
https://github.com/adafruit/Adafruit_NeoPixel/blob/master/examples/simple/simple.ino

maxint-rd, 2025-04-13:
Good to know that the Adafruit library works. Thank you for your testing.

closed 2025-04-24 by yangliangquan