openwch/arduino_core_ch32 #187: CH32X033 support

ITstreet1, 2025-04-02:
Hello, 
I just looked across this repo and didn't find support for this chip. There are some topics in Issues, and it looks like this chip can be programmed with Arduino. If so, which one should I select, as this one is not in the list?

maxint-rd, 2025-04-02:
Hello @ITstreet1 , there is no support in core release 1.0.4 or in the current master branch. But yes, it can be programmed with Arduino. 

Since I bought a bunch of them and wanted to use them with the Arduino IDE, I made the X033 (in TSSOP20 package) work for me. Although perhaps not final, I thought my work could be helpful for others and submitted PR #171 . Read the description of that PR to know the specifics. I didn't have time to make a fool-proof installation, so you may need to put some effort in to make it work for you. Please submit your findings as a comment to that PR or in this topic. 

What I like about the X033 is that the TSSOP20 package has a similar pinout as the CH32V003 but much more memory, allowing it to be used for debugging during development. Just be aware that it is not identical and as you already noted there are some issues that have been submitted already. Good luck!

ITstreet1, 2025-04-24:
Hello,

I want to confirm that your "fixes" work well. Tried just a simple Blink, and it works. It would be nice if this were included in the main branch.

Is there any way to upload a program using a USB?

maxint-rd, 2025-04-24:
Hello @ITstreet1 - thanks fot the feedback. Good to read things work for you as well. I'd also like the various PRs' to be merged. Activity from WCH has been quite low in the past period, but today I saw some issues being closed, so that makes me hopeful for the future. 

In the mean while I've managed to get dual UART working on the X033 and intend to submit the required updates sometime soon. There are some other active community members that have contributed useful PR's. PR #191 was submitted recently and may be helpful to make releases that include more improvements.

I haven't done anything with USB yet, so I can't inform you on that. If I remember well, there was some work done for this repository regarding USB support, but don't know much about it yet. The X035 seems to have a build-in bootloader, so perhaps the X033 has one too. There is some more info in issue #134 and PR #129. The keyword for this seems to be WCHISP.

FYI: I did come across this [X033 dev board](https://github.com/wagiminator/Development-Boards/tree/main/CH32X033F8P6_DevBoard) by Stefan Wagner. who also made a [USB-Knob](https://github.com/wagiminator/CH32X033-USB-Knob) and a [NRF bridge](https://github.com/wagiminator/CH32X033-USB-NRF) for CH32X033 that utilize USB, but I think he doesn't use the Arduino IDE,

maxint-rd, 2025-08-24:
> https://github.com/NgoHungCuong/Arduino_X033
The version has many bugs but it can be downloaded by USB

Very interesting. Good find!

From first looks that core appears to be based on an early version of the CH32 Arduino core; trimmed down for the X033; with added bootloader (using 8kB, sources [here](https://github.com/NgoHungCuong/x033_bootloader)) and board files. I need to take a deeper look and try things out to see how it all works. It may require some installation effort since the board json references a missing xpack release, possibly [this one](https://github.com/xpack-dev-tools/riscv-none-elf-gcc-xpack/releases/tag/v8.2.0-3.1) forked from [here](https://github.com/xpack-dev-tools/riscv-none-elf-gcc-xpack/releases).

For other readers: the repository also includes schematics+pcb for this nice compact development board:
[image]