openwch/arduino_core_ch32 #188: Pull-up Resistor

thevien2507, 2025-04-04:
To establish I2C communication between two CH32V003 microcontrollers, do I need to add external pull-up resistors to the SDA and SCL lines, or do the microcontrollers have internal pull-ups?

maxint-rd, 2025-04-05:
In my experience: Yes. You definitely need one set off pull-up resistors on the I2C bus.
In my I2C experiements I used two 10k resistors. If there are internal pull-ups, they are not enabled hy default and I have not (yet) found the way.

yangliangquan, 2025-04-24:
The pull-up resistor inside V003 has a resistance of 40K. Such a large resistance may cause the voltage of the high level to fail to reach the threshold of the logic high level. It may also cause the rising edge to slow down. Moreover, when there are many devices on the bus, there may be a problem of insufficient driving capability. Therefore, it is recommended to pull up an external resistor with a relatively small resistance, such as about 2K ohms.

maxint-rd, 2025-04-24:
@yangliangquan - thank you for providing such good and thorough information about this (and for closing the issues that are resolved).

thevien2507, 2025-04-25:
@yangliangquan Thanks for the explanation. I’ve actually tried I2C communication between two V003 chips, but it couldn’t send or receive any data. I suspected it was because the internal 40K pull-ups weren’t strong enough, and the signal wasn’t reaching the logic high threshold. After adding external 4.7kΩ pull-up resistors, the communication started working properly. So yeah, using lower-value external pull-ups really helps.