openwch/arduino_core_ch32 #194: CH32V003J4M6 to stops reporting MCU UID after Arduino upload, bricking chip.

beehphy, 2025-04-26:
Uploading a seemingly benign arduino sketch causes CH32V003J4M6 to stop reporting MCU UID. This exact file has been repeated twice, bricking both chips. The circuit is just the chip and a bypass cap with +5V, GND, SWO lines connected WCH-LinkE v1.0.
After upload of INO file the WCH-LinkUtility 2.40 cannot communicate with the chip. The log in in WCH-LinkUtility says:

```
Begin to set chip type...
Failed,the chip type is not matched or status of chip is wrong!
```

 Seems like adding the analogWrite() in the setup function was the only change to cause bricking.

Full Arduino sketch:
```
void TIMx_PWM(u16 ClkPrescaler, u16 Period, u16 PwmCount, u16 Timer, u16 Channel)
{
  TIM_OCInitTypeDef       TIM_OCInitStructure = {0};
  TIM_TimeBaseInitTypeDef TIM_TimeBaseInitStructure = {0};

  TIM_TypeDef *TIMx;
  if      (Timer == 1) {
    TIMx = TIM1;
    RCC_APB2PeriphClockCmd(RCC_APB2Periph_TIM1, ENABLE);
  }
  else if (Timer == 2) {
    TIMx = TIM2;
    RCC_APB1PeriphClockCmd(RCC_APB1Periph_TIM2, ENABLE);
  }
  else {
      return;  //fail, cannot proceed
  }

  TIM_TimeBaseInitStructure.TIM_Period = Period;
  TIM_TimeBaseInitStructure.TIM_Prescaler = ClkPrescaler;
  TIM_TimeBaseInitStructure.TIM_ClockDivision = TIM_CKD_DIV1;
  TIM_TimeBaseInitStructure.TIM_CounterMode = TIM_CounterMode_Up;
  TIM_TimeBaseInit(TIMx, &TIM_TimeBaseInitStructure);

  TIM_OCInitStructure.TIM_OCMode = TIM_OCMode_PWM1;
  TIM_OCInitStructure.TIM_OutputState = TIM_OutputState_Enable;
  TIM_OCInitStructure.TIM_Pulse = PwmCount;
  TIM_OCInitStructure.TIM_OCPolarity = TIM_OCPolarity_Low;
  switch(Channel){
      case 1:
      TIM_OC1Init(TIMx, &TIM_OCInitStructure);
      break;

      case 2:
      TIM_OC2Init(TIMx, &TIM_OCInitStructure);
      break;

      case 3:
      TIM_OC3Init(TIMx, &TIM_OCInitStructure);
      break;

      case 4:
      TIM_OC4Init(TIMx, &TIM_OCInitStructure);
      break;
  }

  TIM_CtrlPWMOutputs(TIM1, ENABLE);
  switch(Channel){
  case 1:
      TIM_OC1PreloadConfig(TIMx, TIM_OCPreload_Disable);
      break;

  case 2:
      TIM_OC2PreloadConfig(TIMx, TIM_OCPreload_Disable);
      break;

  case 3:
      TIM_OC3PreloadConfig(TIMx, TIM_OCPreload_Disable);
      break;

  case 4:
      TIM_OC4PreloadConfig(TIMx, TIM_OCPreload_Disable);
      break;
  }
  TIM_ARRPreloadConfig(TIMx, ENABLE);
  TIM_SelectOutputTrigger(TIMx, TIM_TRGOSource_Update);
  TIM_Cmd(TIMx, ENABLE);
}

void pinSetup(GPIO_TypeDef *port, uint32_t pin, GPIOMode_TypeDef mode) {
    if (pin == 0) printf("Pin setup Bonk\r\n");

    u32 BitMask = 0;
    while ( (1&(pin >> BitMask)) != 0x1 ) {  //solve the pin number based on the mask
        if (BitMask == 20) {
            printf("Pin setup Lost, 0x%x\r\n", pin);
            break;
        }
        BitMask++;
    }

    if ( port == GPIOA_BASE) {
        RCC_APB2PeriphClockCmd(RCC_APB2Periph_GPIOA , ENABLE);
    }
    if ( port == GPIOC_BASE) {
        RCC_APB2PeriphClockCmd(RCC_APB2Periph_GPIOC , ENABLE);
    }

    GPIO_InitTypeDef  GPIO_InitStructure = {0};  //blank structure
    GPIO_StructInit(&GPIO_InitStructure);
    GPIO_InitStructure.GPIO_Pin = pin;
    GPIO_InitStructure.GPIO_Mode = mode;
    GPIO_Init(port, &GPIO_InitStructure);  //make active
}

void setup() {
  Serial.begin(115200);

  analogWrite(PC4,128);  //this line seems to be the problem
  //I was having prblems getting the PWM output going. 
  //I added the analogWrite hoping to poke the pin configuration into opperation

  pinSetup(GPIOC, GPIO_Pin_4 ,GPIO_Mode_AF_PP);
  TIMx_PWM(1,200,20,TIM1,4);
}

int i = 0;

void loop() {
  Serial.print(i++);
  delay(100);
}
```

maxint-rd, 2025-04-26:
This is the SOP8 package where pins have multiple functions. My guess is that using serial removed the SWIO function. See [this comment](https://github.com/openwch/arduino_core_ch32/issues/143#issuecomment-2345545212)

beehphy, 2025-04-27:
The PWR erase function was working before uploading the mentioned code. 
Here is the log result of attempting to erase the chip after.
```
22:08:42:311>> Succeed to connect with WCH-Link!
22:08:54:727>> Begin to clear all code flash...
22:08:57:235>> Failed!
```

maxint-rd, 2025-04-27:
Bummer....I'm out of ideas. Maybe @yangliangquan or @DeqingSun have ideas what to do?

yangliangquan, 2025-04-28:
@beehphy I checked your code and found that you used `PC4 `instead of `PC_4 `in `analogWrite(PC4, 128); //this line seems to be the problem. `During the table lookup process, it will be found that PC4 is not in PinMap_TIM, so it may not be able to output a PWM waveform. Then I tried to download your code and didn't find any fatal errors that would cause the debugging port of the chip to be completely closed and unable to be re - connected through PWR power - down erasure. I have the following possible guesses about the situation you encountered. Maybe the Link you are using is not upgraded to the latest version. You can try to hold down the IAP button on the Link while connecting it to the computer to **update your Link**. It's also possible that the **cable connecting the Link to the chip is too long**. If so, please shorten the connection and try again.

beehphy, 2025-04-29:
@yangliangquan  Excellent Find on the pin misnaming. I will explore this as the cause of the lockup. It is strange the compiler would not catch the wrong name with an error.

I have recovered the unit seemingly bricked units, but I had to use a different programmer to do so. 

I was failing to recover the device using the Muse Labs programmer found here:
https://www.tindie.com/products/johnnywu/wch-linke-debugger/
After confirming most recent FW it still gives this message when running the PWR erase cycle:
```
11:14:00:655>> Connected RISC-V mode WCH-Link Cnt:0
11:16:40:016>> Succeed to connect with WCH-Link!
11:16:47:898>> Begin to clear all code flash...
11:16:49:400>> Failed!
11:17:06:186>> Begin to set chip type...
11:17:06:246>> Failed,the chip type is not matched or status of chip is wrong!
```

I switched to the WCH-LinkE-RO-1v3 and was able to recover the devices with PWR erase cycle:
https://www.wch-ic.com/products/WCH-Link.html

closed 2025-04-29 by beehphy

DeqingSun, 2025-04-30:
@beehphy Hardware side, WCH-LinkE-RO-1v3 has 2 of the CH217K chip than can control on/off of 5V and 3.3V. Muse Labs programmer does not have any part to do so. That is why WCH-LinkE-RO-1v3 can do a power cycle.