openwch/arduino_core_ch32 #197: Release v1.0.5 with wchisp

delfer, 2025-05-21:
Release it pls

rambros3d, 2025-05-25:
Its been a while since the last update.

@TianpeiLee @openwch

DeqingSun, 2025-05-30:
For anyone just want wchisp, you can try this
https://github.com/DeqingSun/arduino_core_ch32/releases/download/arduinoPack/package_ch32v_index_add_ch573.json

boda-akos, 2025-06-15:
[image]  Hello, using the CH32V006E4R6 24pin  , installed the link in the previous post,  compile OK but can not upload. Clock is 24Mhz internal (correct), WCH-LinkE and my board is OK using MounRiver Studio. Please give some clue, thanks . , , Waiting for the 1.0.5 release thou...

maxint-rd, 2025-06-15:
Hello @boda-akos - 
To enable uploading and interactive debugging from the Arduino IDE you need to update OpenOCD using the copy from MRS. I've published some instructions [here](https://github.com/maxint-rd/arduino_core_ch32/blob/support-CH32V002F4-CH32V006F8/variants/CH32VM00X/README.md#updating-openocd).

To improve support for the 24 pin package you may find the inspiration in my my recent PR #200, which adds a board definition for the 20-pin TSSOP package.

boda-akos, 2025-06-23:
> Hello @boda-akos - Can you give some details about your test setup?
> I haven't tried anything SPI related yet, but when I have some time available, I can try to reproduce your issue.
>
> What package of the V006/V003 were you using? I have TSSOP20 versions of V003, V002 and V006 available for testing.
>
> What flash chip did you connect to the V006? I can try to find a similar flash chip in my collection (don't know which ones I have).
>
> In the .ino I see this pinout for the V003:
>   CS:   PA2 (003 is PC4)
>   MOSI: PA7 (003 is PC6)
>   MISO: PA6 (003 is PC7)
>   SCK:  PA5 (003 is PC5)
>
> I assume you used the ones listed between parenthesis for V003. The V003 pinout diagram shows PC1 as NSS (CS) pin. The example uses A2 (PC4). Usually the CS pin can be any random GPIO pin.
> Do you have any pull-ups connected? If so, what value?
>
> Is there anything else specific to your setup? What voltage are you using? Did you modify the example code in any way?

Hello maxint-rd
Thanks in advance for your time. I send some photos about the things I use. I started 3 weeks ago with the 003 but soon filled up the 16k. With that chip, most libraries work well. Also , the SPI example uses the pins what is on your answer as "003 is" etc. 
I have 28 pin and 20 pin versions of the 006 which I find a very promising chip. By the way, I am a hobbyist, no more, never studied coding professionally. Please see my page here :   https://www.thingiverse.com/akosboda/designs
Allright, about the subject. I2C was a success on the 006. So was the timer interrupt. I wrote SPI driver for this TFT ST7735 bit-banging, and also for other types. It is fast enough (even had to delay 1usec when writing) but I can not use the Adafruit GFX library this way I think - and I'd love to. This is why I tried the SPI library, it is OK on the 003 but on the 28 pin chip no success at all. The default pins are not available on the 28 pin - PC7 is N/A , PC5 is RST. So i did AFIO->PCFR1 |= 0x02; this to remap the pins and the register is alive if I read it back I see the value so there is also clock, but the outputs remain floating. No other pins move when execute SPI.transfer(); don't know what I miss still.
I am not using pull-ups. The output CLS MOSI etc is clearly floating, see it on the oscilloscope. CS works of course. 
I tried to modify the default pins in the libraries (attached, SPI in the end) with remap together but no result. The .ino file was modified in every possible way. So I know that the loop code does run , can read and verify / Serial.print registers etc. There everything was looking good.
Now based on your question or idea, I will try the same procedure on the 20 pin chip which looks better, no conflicting pins. Will write to you about the result.
Kind regards,
Akos Boda

boda-akos, 2025-06-23:
Modified the PeripheralPins.c the following way corresponding to

 // Partial Remap 2: NSS/PC4, SCK/PD2, MISO/PB3, MOSI/PD3  

pin_SetV32AFPin(AFIO_PartialRemap2_SPI1_ENABLE);    

#ifdef SPI_MODULE_ENABLED
WEAK const PinMap PinMap_SPI_MISO[] = {
  //{PC_4, SPI1, CH_PIN_DATA(CH_MODE_INPUT, CH_CNF_INPUT_FLOAT, 0, AFIO_NONE)},
  {PB_3, SPI1, CH_PIN_DATA(CH_MODE_INPUT, CH_CNF_INPUT_FLOAT, 0, AFIO_NONE)},
  {NC,   NP,   0}
};
#endif

#ifdef SPI_MODULE_ENABLED
WEAK const PinMap PinMap_SPI_SCLK[] = {
  //{PC_5, SPI1, CH_PIN_DATA(CH_MODE_OUTPUT_50MHz, CH_CNF_OUTPUT_AFPP, 0, AFIO_NONE)},
{PD_2, SPI1, CH_PIN_DATA(CH_MODE_OUTPUT_50MHz, CH_CNF_OUTPUT_AFPP, 0, AFIO_NONE)},
  {NC,   NP,   0}
};
#endif

#ifdef SPI_MODULE_ENABLED
WEAK const PinMap PinMap_SPI_SSEL[] = {
  //{PC_1,  SPI1, CH_PIN_DATA(CH_MODE_OUTPUT_50MHz, CH_CNF_OUTPUT_AFPP, 0, AFIO_NONE)},
  {PC_4,  SPI1, CH_PIN_DATA(CH_MODE_OUTPUT_50MHz, CH_CNF_OUTPUT_AFPP, 0, AFIO_NONE)},

  {NC,    NP,   0}
};
#endif

closed 2026-10-28 by medtty