openwch/arduino_core_ch32 #202: CH32V003: Once write a sketch, can't write a sketch again

hidori, 2025-06-20:
Hi,

In CH32V003, PD1 and SWIO are assigned to the same pin, and whether it behaves as GPIO or SWIO is determined by the setting of the built-in register of muc.

In $/system/CH32V00x/USER/system_ch32v00x.c, PD1 is configured to act as gpio as shown below.

```cpp
    RCC->APB2PCENR |= RCC_AFIOEN;
    AFIO->PCFR1 |= (1<<15);
```

Therefore, once you write a sketch to CH32V003, you will never be able to write a sketch to it again.

When developing using the Arduino IDE, most people will likely write sketches multiple times, so this behavior is unacceptable.

I would like this corrected.

Thanks.

maxint-rd, 2025-06-20:
Hmm.. I'm not sure if I understand your issue. 
For details see [my comment](https://github.com/openwch/arduino_core_ch32/pull/203#issuecomment-2992676992) in PR #203

maxint-rd, 2025-07-03:
Hello @hidori , is your issue resolved? If so, please close it...