openwch/arduino_core_ch32 #206: Restore SWIO communication on CH32V003 CH32V006

boda-akos, 2025-06-30:
Connect WCH-LinkE as usual. Run WCH-Link. Hold reset pin PD7 or PD5 to GND while Query Chip Info 2-3 times while holding reset on GND.  Release reset pin, Run the Query again and note that there was communication. Connect reset to GND and run Query 2-3 times again, then leave WCH-Link, switch to Arduino IDE and upload your program with success after that.

maxint-rd, 2025-06-30:
Hello @boda-akos - The regular way to restore SWIO communication is to use the Erase using Power Off function of the WCH link utility. Was that method not working for you?

boda-akos, 2025-07-01:
I did not know about this method. Still on the learning path ...

ewpa, 2025-08-02:
> Hello [@boda-akos](https://github.com/boda-akos) - The regular way to restore SWIO communication is to use the Erase using Power Off function of the WCH link utility. Was that method not working for you?

I found that power-off erase function does not work for CH32V006 or CH32V002.  I'd love to revive my growing pile of dead chips.  Any hints?

maxint-rd, 2025-08-02:
> I found that power-off erase function does not work for CH32V006 or CH32V002. I'd love to revive my growing pile of dead chips. Any hints?

A pile of dead V006/V002? What did you do?
Have you updated your LinkE to the latest version?
When you connect them on a bare adapter board using only GND/5V/SWIO, do you get any response? (note: use short wires and don't use anything else, no caps or other passives either)

ewpa, 2025-08-02:
For V002, I'm converting from 8-pin V003 and needed the SWIO pin as output.  I also seemed to brick one before I updated LinkE to 2.15.  Waiting for some DIP-8 adapters to arrive in the mail.

ewpa, 2025-08-02:
I saw nothing being read back, as if the chip were missing.  I have since reworked that board, but still have the chip ready for when my DIP-8 breakout boards arrive.  I've also lined up power, ground and SWIO, so it can plug straight into the LinkE, no wires.

boda-akos, 2025-08-02:
> I found that power-off erase function does not work for CH32V006 or CH32V002. I'd love to revive my growing pile of dead chips. Any hints?

Hello, I only had one or two chips and revived them with the method as described, not with the Erase using Power Off. Sorry, but I can not help more. Since then, had no bricked chips.

closed 2025-08-02 by boda-akos

ewpa, 2025-08-05:
Upgraded LinkE firmware from 2.15 to 2.18.  Power-off erase functions on working V002 (though I did not try making the SWIO pin as output).  For the bricked V002, no result.  Always get "the chip type is mismatched or status of chip is wrong".  My earlier toolchain (hardware and software) was set for V003 before I started all this, so wonder if this somehow caused read and write protect to be enabled?

I cannot try the PD7/PD5 method since those pins are not available on the 8-pin chip.

However, for now I am moving back to V003.  I'm having a lot of other (unrelated to this) issues with porting to V002.

boda-akos, 2025-08-12:
Well again I had a bricked CH32V006 restored. Power cycle did not work, because after switching to external crystal I did this stupidity : "Disable HSI if HSE is stable  RCC->CTLR &= ~RCC_HSION;" this bricked SWD. But, success story follows. To begin with, we can exclude any hardware fault, there was none. I had to run WCH Link-Utility V2.50, Query Chip Info while holding the Reset pin  of the CH32 to GND, and do this 3 times. Releasing the Reset the Query Chip Info was working already. After this I switched to Arduino IDE and repeated the program download the same way , holding the Reset to GND two times, releasing the third time, and see miracle the upload is working since then.

boda-akos, 2025-08-12:
Of course the first instruction executed after this was RCC->CTLR |= RCC_HSION; to restore the internal 24MHz clock.

ewpa, 2025-08-12:
> BTW. For programming/debugging the V002/V006 from the Arduino IDE it's also necessary to update to a newer version of WCH's variant of OpenOCD (see also [this readme](https://github.com/maxint-rd/arduino_core_ch32/blob/support-CH32V002F4-CH32V006F8/variants/CH32VM00X/README.md)).

Got my QFN-12 breakout boards now so I will try this OpenOCD.  My development lifecycle does not include Windows or GUI.  I've been using `minichlink` patches.  I've not pushed these patches to GitHub.

> Indeed. The CH32V002J4M6 (SOP8) has no pin for RST. Somewhere I read a post where someone revived his chip by connecting a crystal. For SOP8 (V003 and V002) the Crystal pins X1 and X0 are on pins 1 and 3. According the reference manual such configuration requires crystal and two load capacitors. (I've not used crystals myself other than on the V003 EVT board).

The crystal (I used 16 MHz) made no difference.

Only programming software that responds in any way is `minichlink`.
```
link error, nothing connected to linker (4 = [81 55 01 01]).  Trying to put processor in hold and retrying.
Unknown chip type.  Report as bug with picture of chip.
marchid : dc68d843
HARTINFO: 002120f4
Could not setup interface.
```
The above `hartinfo` and `marchid` both look good according to QingKeV2 manual.  Strange.

> What kind of other issues do you have? I've made a bunch of improvements for V002. Most of them I shared in PR [#200](https://github.com/openwch/arduino_core_ch32/pull/200). I have some projects that I wanted to use on V002/V006 for better ADC and for debugging. It resulted in the need to make those improvements. Maybe your issues are related, (I've not tested everything yet, so perhaps I still need to encounter your V002 issues). Can you perhaps open a new issue for those V002 issues or add a comment in PR [#200](https://github.com/openwch/arduino_core_ch32/pull/200) listing them?

In summary, no response from V002 MCU (e.g. blink sketck) except when using the V003 target.  And then, MCU runs at a different speed every time I reset it.  Let me try OpenOCD and the QFN-12 chips.

ewpa, 2025-08-12:
> it's also necessary to update to a newer version of WCH's variant of OpenOCD (see also [this readme](https://github.com/maxint-rd/arduino_core_ch32/blob/support-CH32V002F4-CH32V006F8/variants/CH32VM00X/README.md)).

Hi @maxint-rd, where can I find the openocd source code so I can compile and try this?  Is it patched for WCH?

maxint-rd, 2025-08-13:
Hmmm... that's something we may need to look into. This evening I was working on porting a somewhat larger project to the V006 (using my latest code) when I encountered similar behavior. 
Stepping through setup() worked fine, but going into the loop() ended me hanging...
I then tried the simple standard Blink example and could not reproduce that behavior. 

Maybe a peripheral is misbehaving and causing the hard fault when debugging? I have no clue (yet). For now I'll try to work around it and let you know if I find out more...

maxint-rd, 2025-08-13:
Update: I found the cause for the hard-fault that I encountered. In a library a datapin was switched from output to input with pull-up:
```
  // Pull-up on
  pinMode(dataPin, INPUT);
  digitalWrite(dataPin, HIGH);
```

After reading data the pin was switched back:
```
  // Pull-up off
  pinMode(dataPin, OUTPUT);
  digitalWrite(dataPin, LOW);
```

As you can see, this is not making a nice sandwich. Slowly stepping through would not trigger the fault, but having a quick go (or step over) would. 
Although working fine on other platforms, it appears the V006 is more sensitive to the correct order:
```
  // Pull-up off
  digitalWrite(dataPin, LOW);
  pinMode(dataPin, OUTPUT);
```

After changing said order, the debugger would no longer hang in the hard fault handler. Perhaps your situation is different, but knowing that "incorrect" use of pinMode()/digitalWrite() can lead to such error is possibly a clue. I can imagine there are other, similar circumstances where use of some peripheral can trigger that hard fault.

boda-akos, 2025-08-14:
Why not to use direct register access. Here is how - from chinese AI :

Avoiding pinMode() and digitalWrite() for Direct Register Access on CH32V002
For optimal performance on the CH32V002 microcontroller (a RISC-V MCU), you can directly manipulate the GPIO registers instead of using the Arduino-style pinMode() and digitalWrite() functions. Here's how to do it:

GPIO Register Basics
The CH32V002 has several important registers for each GPIO port (GPIOA, GPIOB, etc.):

CFGLR - Configuration register low (pins 0-7)

CFGHR - Configuration register high (pins 8-15)

OUTDR - Output data register

INDR - Input data register

BSHR - Bit set/reset register

BCKR - Bit clear register

Direct Register Manipulation Examples
1. Setting Pin Mode (Alternative to pinMode())
c
// Set PA1 as output push-pull (mode 0b00, CNF 0b00)
GPIOA->CFGLR &= ~(0b1111 << (4 * 1));  // Clear existing settings
GPIOA->CFGLR |= (0b0000 << (4 * 1));   // 00: General output, 00: Push-pull

// Set PA2 as input floating (mode 0b01, CNF 0b01)
GPIOA->CFGLR &= ~(0b1111 << (4 * 2));
GPIOA->CFGLR |= (0b0101 << (4 * 2));   // 01: Input, 01: Floating
2. Writing Digital Output (Alternative to digitalWrite())
c
// Set PA1 high
GPIOA->BSHR = (1 << 1);  // Bit Set

// Set PA1 low
GPIOA->BCKR = (1 << 1);  // Bit Clear

// Or alternatively:
GPIOA->OUTDR |= (1 << 1);   // Set high
GPIOA->OUTDR &= ~(1 << 1);  // Set low
3. Reading Digital Input
c
uint8_t value = (GPIOA->INDR >> 2) & 1;  // Read PA2

boda-akos, 2025-08-15:
Off topic, sorry, but one of my favorites : run the Blink example, with digitalWrite() the time for this loop is 4171 ms, direct write 334 ms, a 12.5*  speed increase. DeepSeek tells you why.
 
void loop()
 {
tim=millis();
for (uint32_t i=0; i<1000000; i++) {
  //digitalWrite(PC0, HIGH);  // turn the LED on (HIGH is the voltage level)
  //digitalWrite(PC0, LOW);   // turn the LED off by making the voltage LOW
GPIOC->OUTDR |= (1 << 1);   // Set high
GPIOC->OUTDR &= ~(1 << 1);  // Set low
                                                   }
 Serial.println(millis()-tim);
}

maxint-rd, 2025-08-15:
Off topic indeed, interesting nevertheless...
Last months I was focusing on getting LTO working to get flash size down (for V002/V003) and [once working](https://github.com/openwch/arduino_core_ch32/pull/215) I noticed it also improved execution speed.

I tested your sketch on the V006 and the V003, both running @ 48MHz HSI. 
These are my timings and binary sizes with various optimization options:

|MCU | Optimization option | Using registers| Using digitalWrite|
|----|--------|-------|----|
|V006 |Debug (-Og)  |625ms (7572 bytes)|5943ms (7752 bytes)|
|V006 |Smallest (-Os default) |438ms (6176 bytes)|5943ms (6352 bytes)|
|V006 |Smallest (-Os) with LTO |375ms (5172 bytes)|1501ms (5200 bytes)|
|V006 |Fastest (-O3) |438ms (8372 bytes)|5756ms (8540 bytes)|
|V006 |Fastest (-O3) with LTO |375ms (7824 bytes)|**312ms** (7800 bytes)|
|V003 |Debug (-Og)  |479ms (5604 bytes)|4295ms (5776 bytes)|
|V003 |Smallest (-Os default) |334ms (4840 bytes)|4004ms (5008 bytes)|
|V003 |Smallest (-Os) with LTO |334ms (3808 bytes)|1126ms (3836 bytes)|
|V003 |Fastest (-O3) |334ms (6212 bytes)|4087ms (6372 bytes)|
|V003 |Fastest (-O3) with LTO |334ms (5816 bytes)|**292ms** (5780 bytes)|

Conclusion: 
- Using registers is much faster than using digitalWrite (as we already knew), *except for option Fastest (-O3) with LTO (!)*
- When using registers, the options fastest and smallest are just as fast (in this example). 
- Compiling with LTO is smaller and fastest. Using digitalWrite is much faster with LTO than without. 
- default option is smaller but slower.
- Debug is large and slow (as expected), but not as large as option Fastest.
- V003 is faster than V006 (and also uses less flash) and has same speed regardless the optimization option

Most interesting I think is that when using registers on the V003 the timings are the same, whereas for V006 using LTO seems to make a difference. digitalWrite compiled option Fastest (-O3) with LTO seems faster than using registers! Perhaps you can try #215 and check these remarkable findings?
(Both for V003 and V006 the produced binary for -O3 with LTO is smaller when using digitalWrite. I guess we need to look at the disassembled ELF code to see how that happens).

BTW. I noticed the led was brightest for the slowest options, dimmed at medium speed and off at fastest options. I didn't check the signal with a logical analyzer or measured the frequencies.

Note: for the V002/V003 I made some more optimizations to reduce used flash size than for V006.

ewpa, 2025-08-16:
> I think I will work under that assumption and turn my attention to getting the board definitions set when I next get a chance.

Finally got V002 8-pin and 12-pin packages to work with the VM00x core.  Looking at the `sp` register in `libc_init_array` assembly I see it is set to 8kB (0x20002000) but the RAM is only 4kB (0x20000000 to 0x20001000) on V002.  It turns out that in `system/CH32VM00X/SRC/Ld/Link.ld` RAM is hard-coded to _always_ be 8kB...

ewpa, 2025-08-16:
> Aren't you using the code of my PR?

No, there were many other functional changes in the PR.  I just quickly tried and see this:
```
Failed uploading: cannot execute upload tool: fork/exec {runtime.tools.openocd-2.3.0.path}/bin/openocd: no such file or directory
```

maxint-rd, 2025-08-16:
There are [installation instructions](https://github.com/maxint-rd/arduino_core_ch32/blob/support-CH32V002F4-CH32V006F8/variants/CH32VM00X/README.md#fresh-installation-with-support-for-ch32vm00x-v002v006etc) mentioned at the top of the PR. It describes how I performed a fresh installation on a Windows 11 VM. It mentions that uploading from the IDE to V002/V006 requires [updating OpenOCD](https://github.com/maxint-rd/arduino_core_ch32/blob/support-CH32V002F4-CH32V006F8/variants/CH32VM00X/README.md#updating-openocd).

Unfortunately I am in no position to provide an easy, instantly working Arduino core, so I try my best to make my own development environment work well and try to share my improvements wherever I can. It requires some effort to apply the many useful PR's onto the latest master code. Sometimes some manual changes are needed to make it all work.

ewpa, 2025-08-16:
I don't have Windows.  I made a partial revert of commit 2b0263c1 and OpenOCD calls work now on your branch.

maxint-rd, 2025-08-16:
Nice! Congratulations on getting it working in your environment. I appreciate your participation in the discussion and hope it will help others in their use of the CH32 Arduino core. Wish you good luck in your projects and hopefully you don't get any more bricked chips. Have fun!

ewpa, 2025-08-16:
Thanks @maxint-rd, I was hoping the V002 was a drop-in replacement for the V003 but have learned a lot.

maxint-rd, 2025-08-16:
You're welcome. I've also learned a lot.

For other readers: Pin-out and flash size may be the same, but there are differences. (e.g. more RAM, no OPA, higher ADC resolution). I see the VM00X series more as an upgrade to the V003. (Find [here](https://github.com/maxint-rd/arduino_core_ch32/blob/support-CH32V002F4-CH32V006F8/variants/CH32VM00X/README.md#feature-comparison-with-v003-tssop20) the main features of V003 compared to the VM00X series).

For my CH32 Arduino projects I added TSSOP20 versions of V002 and V006 to my collection. The V006 is nice processor during development as its larger flash allows for extensive serial messages and real time debugging. When done I can easily strip serial and compile with LTO to fit it into the cheaper V002/V003, without too many changes,

BTW. I think there are still advantages for using the V003 vs. V002, such as a smaller flash memory usage and a bit more speed. See also the table comparing speed/flash size [posted above](https://github.com/openwch/arduino_core_ch32/issues/206#issuecomment-3192803582).