openwch/arduino_core_ch32 #208: MacOS Apple Silicon Support

zoul0813, 2025-07-03:
Arduino IDE Version 2.3.6
CH32 MCU EVT Boards 1.0.4
MacOS Sequoia 15.5 (M1 Pro)

When installing on MacOS with Apple Silicon, `brew install libusb` places the dylib in `/opt/homebrew/Cellar/libusb/1.0.29/lib/libusb-1.0.0.dylib` which does not appear to be in the search path

When symlinking this to /usr/local/lib/libusb-1.0.0.dylib so it can be found, it then errors saying that the dylib is for 'arm64' and it's looking for 'x86_64'

```
dyld[47342]: Library not loaded: /usr/local/opt/libusb/lib/libusb-1.0.0.dylib
  Referenced from: <C79B1DDB-9DAE-30AE-9474-2E146265EC5A> ~/Library/Arduino15/packages/WCH/tools/openocd/1.0.0/bin/openocd
  Reason: tried: 
  '/usr/local/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), 
  '/System/Volumes/Preboot/Cryptexes/OS/usr/local/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), 
  '/usr/local/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), 
  '/usr/local/lib/libusb-1.0.0.dylib' (mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64')), 
  '/usr/lib/libusb-1.0.0.dylib' (no such file, not in dyld cache)
Failed uploading: uploading error: signal: abort trap
```

maxint-rd, 2025-07-03:
Hello @zoul0813 , I believe this issue is the same as issue #173. Perhaps you can find some help in the responses and comments to that issue.

(BTW. I don't have a Mac, but have read most issues in the past and try to help out by pointing people to those issues).

zoul0813, 2025-07-04:
@maxint-rd it is related, however, that issue is closed, and the issue still exists.

I attempted to follow the suggestions in that thread, and was unable to get a working setup.  I downloaded the [MRS_Toolchain_MAC_V210.zip](https://www.mounriver.com/download) and copied `MRS_Toolchain_MAC_V210/OpenOCD/OpenOCD/bin/openocd` to `~/Library/Arduino15/packages/WCH/tools/openocd/1.0.0/bin/openocd`, this results in an error trying to load the libusb dylib from an unknown location

```
dyld[7638]: Library not loaded: @loader_path/../../../Others/CommunicationLib/default/libusb-1.0.0.dylib
  Referenced from: <6EA6E937-64C3-4941-BD93-0EA5C9F6A833> ~/Library/Arduino15/packages/WCH/tools/openocd/1.0.0/bin/openocd
  Reason: tried: '~/Library/Arduino15/packages/WCH/tools/openocd/1.0.0/bin/../../../Others/CommunicationLib/default/libusb-1.0.0.dylib' (no such file)
Failed uploading: uploading error: signal: abort trap
```

If I then symlink the libusb dylib installed via homebrew to this location, I get the following error

```
dyld[7982]: Library not loaded: @loader_path/../../../Others/CommunicationLib/default/libusb-1.0.0.dylib
  Referenced from: <6EA6E937-64C3-4941-BD93-0EA5C9F6A833> ~/Library/Arduino15/packages/WCH/tools/openocd/1.0.0/bin/openocd
  Reason: tried: '~/Library/Arduino15/packages/WCH/tools/openocd/1.0.0/bin/../../../Others/CommunicationLib/default/libusb-1.0.0.dylib' (code signature in <B0D386D3-DE68-3AC0-97AA-3AC07386EDE9> '~/Library/Arduino15/packages/WCH/tools/Others/CommunicationLib/default/libusb-1.0.0.dylib' not valid for use in process: mapping process and mapped file (non-platform) have different Team IDs)
Failed uploading: uploading error: signal: abort trap
```

I did manage to find another thread, which suggested doing the following 

```
codesign --force -s - openocd
brew install hidapi
```

after doing this, I was able to get the Arduino IDE to build, and upload the code ... after going through some Apple security prompts as I removed the signature from openocd, so I had to allow it to be run as an unsigned app ... I also had to symlink the homebrew libhidapi dylib to the same location (Others/CommunicationLib/default) as well.

After this, I get this following in the IDE when uploading

```
Sketch uses 4728 bytes (28%) of program storage space. Maximum is 16384 bytes.
Global variables use 396 bytes (19%) of dynamic memory, leaving 1652 bytes for local variables. Maximum is 2048 bytes.
Open On-Chip Debugger 0.11.0+dev-02415-gfad123a16-dirty (2024-11-11-17:07)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'sdi'
Warn : Transport "sdi" was already selected
Ready for Remote Connections
Info : WCH-LinkE  mode:RV version 2.10 
Info : wlink_init ok
Info : clock speed 6000 kHz
Info : [wch_riscv.cpu.0] datacount=2 progbufsize=8
Info : [wch_riscv.cpu.0] Examined RISC-V core; found 1 harts
Info : [wch_riscv.cpu.0]  XLEN=32, misa=0x40800014
[wch_riscv.cpu.0] Target successfully examined.
Info : starting gdb server for wch_riscv.cpu.0 on 3333
Info : Listening on port 3333 for gdb connections
** Programming Started **
Info : device id = 0xca91abcd
** Programming Finished **
** Verify Started **
** Verified OK **
```

This appears to work, but this was a lot of steps required to get this far ... providing static linked arm64 binaries would make this "plug and play"

* Download 500MB MRS Toolchain
* Copy `MRS_Toolchain_MAC_V210/OpenOCD/OpenOCD/bin/openocd` to `~/Library/Arduino15/packages/WCH/tools/openocd/1.0.0/bin/openocd`
* Create `~/Library/Arduino15/packages/WCH/tools/Others/CommunicationLib/default`
* `ln -s /opt/homebrew/Cellar/libusb/1.0.29/lib/libusb-1.0.0.dylib ~/Library/Arduino15/packages/WCH/tools/Others/CommunicationLib/default/libusb-1.0.0.dylib`
* `ln -s ln -s /opt/homebrew/Cellar/hidapi/0.15.0/lib/libhidapi.0.dylib ~/Library/Arduino15/packages/WCH/tools/Others/CommunicationLib/default/libhidapi.0.dylib`
* `codesign --force -s - ~/Library/Arduino15/packages/WCH/tools/openocd/1.0.0/binopenocd` (confirm prompts)
* attempt to upload in arduino ide, confirming use of unsigned code and providing admin password (may also require opening settings and going to security/privacy and saying "allow" on the openocd binary)

Hopefully others with the same issue find this workaround useful

maxint-rd, 2025-07-04:
Hello @zoul0813 - thanks for your elaborate post and congratulations with getting something to work! 
BTW: if you ever have problems programming the newer MV00X series (eg. CH32V006), then be aware that those require an update of the LinkE programmer and a newer version of OpenOCD. (I obtained the 2.3 Windows version from a fresh installation of MounRiver Studio).

Without a Mac I can't tell you which method would work for me, but I will gladly refer others with a similar problem to both your posts here and to the ones in issue #173. 

@ Other Mac users: if you had problems and got it working using the method above or something else, please add a comment here to inform others about what worked (and what didn't).