openwch/arduino_core_ch32 #209: When I installed version 1.0.4 of Windows, the IDE promptly popped up with an error message:

RegularSource, 2025-07-06:
When I installed version 1.0.4 of Windows, the IDE promptly popped up with an error message,However, the previous version was without any problems:
What puzzles me is that since I'm using the Windows system, why do I need to install Linux packages?
平台安装失败：'CH32 MCU EVT Boards:1.0.4' 13 INTERNAL: Cannot install tool WCH:beforeinstall@1.0.0: extracting archive: Create link C:\Users\Pc\AppData\Local\Arduino15\tmp\package-3677301233\wch_beforeinstall_linux-1.0.0\libhidapi-hidraw.so: symlink libhidapi-hidraw.so.0.0.0 C:\Users\Pc\AppData\Local\Arduino15\tmp\package-3677301233\wch_beforeinstall_linux-1.0.0\libhidapi-hidraw.so: A required privilege is not held by the client.

maxint-rd, 2025-07-07:
I'm not sure about that message. I don't think you actually need that package. Recently I performed a fresh installation of 1.0.4 on a Windows 11 VM. and I don't recall such message. Important thing is to **run the IDE as Administrator** when installing it.

--------

For the support of V002/V006 in PR #200, I wrote in [the readme](https://github.com/maxint-rd/arduino_core_ch32/edit/support-CH32V002F4-CH32V006F8/variants/CH32VM00X/README.md) this guide of how I did that fresh installation:

### Fresh installation with support for CH32VM00X (V002/V006/etc) ###

1. Install [latest WCH Link Utility](https://www.wch.cn/downloads/WCH-LinkUtility_ZIP.html), update LinkE device when prompted. Test connection to Link Utility and CH32 chip using Query Chip Info. (This step installs drivers and support for CH32V006 and other newer chips).
2. Install latest Arduino IDE (v2.3.6).
3. Run Arduino IDE as administrator, Add [board json](https://github.com/openwch/board_manager_files/raw/main/package_ch32v_index.json) to preferences. Install WCH CH32 core v1.0.4. Exit IDE.
4. Download [zip of latest master branch](https://github.com/openwch/arduino_core_ch32/archive/refs/heads/main.zip). Unzip and copy/overwrite all contents to this location: 
   - `C:\Users\<username>\AppData\Local\Arduino15\packages\WCH\hardware\ch32v\1.0.4`
5. Optional: review [unmerged PRs](https://github.com/openwch/arduino_core_ch32/pulls) for possibly required fixes and manually copy changed files.
6. Optional: for uploading to new chip: [update OpenOCD](https://github.com/maxint-rd/arduino_core_ch32/edit/support-CH32V002F4-CH32V006F8/variants/CH32VM00X/README.md#updating-openocd) obtained from MounRiver Studio.
7. Restart IDE. If new boards are not visible: [clear IDE application cache](https://github.com/maxint-rd/arduino_core_ch32/edit/support-CH32V002F4-CH32V006F8/variants/CH32VM00X/README.md#clearing-arduino-2x-ide-application-cache).
8. Select CH32 board. Test uploading simple serial example. Test debugging.

medtty, 2026-11-07:
Fixed in 1.0.5: the Boards Manager no longer gives Windows the Linux setup files that failed to unpack. Please install 1.0.5; older versions still use the old files

closed 2026-11-07 by medtty