openwch/arduino_core_ch32 #212: Support for CH32X035C8T6

Sod-Almighty, 2025-07-09:
Hi. How would I go about adding support for the CH32X035C8T6? There is already support for the CH32X035G8U, and the only real difference is in the number of pins: The G8 has 28 pins while the C8 has 48.

Would it be fairly straightforward to add this chip? What would I need to do?

Thanks.

maxint-rd, 2025-07-09:
Hello @Sod-Almighty - You would need to add a variant based on the existing definition. When I did the same to add the TSSOP20 variant of the CH32V006, I found it fairly straightforward. All I basically had to do was make a copy of the existing variant and change the copy. You can have a look at PR #200 to see what my changes were and where I made them. Please note that that PR is a bit broader as it also contains the V002 variant and some other changes (such as enabling the IDE menu for peripheral selection).

Next to matching the number of pins I also paid attention to the Arduino pin-numbering, since I wanted the TSSOP20 V006 variant numbers to match with the TSSOP20 V002 and  V003. 

Unfortunately I don't have any X035 in my collection, so I can't offer you any direct assistance or test your definition. In PR #171 you can see the changes I made to support the CH32X033, based on the existing X035 board definition. Good luck!

maxint-rd, 2025-07-09:
> (On a related note: What would happen if I tried to program the C8 while telling the Arduino IDE I'm actually using a G8? Would it work fine, just without the extra pins? I know it's not really a solution, but perhaps I'd at least be able to test the chip and make sure it's not a dud?)

Forgot to answer: in general it should be working, but indeed not with the extra pins. Simple sketches should be no problem. I would start with a simple Blink example and then test other features such as serial and other peripherals. Easiest usage in combination with the datasheet is to use the CH32 pin names (eg. PA1) instead of their (unknown) Arduino pin number.

Note that for many other CH32 board definitions only a few of the packages have a board definition. The CH32V003 for example only has matching support for the TSSOP20 package, while it comes in many other (mostly smaller packages). Same for most other definitions.

maxint-rd, 2025-07-09:
> I'd at least be able to test the chip and make sure it's not a dud?)

Additional remark: my first test is always to first use the WCH LinkE programmer with the WCH Link Utility. If the chip is properly recognized chances are it will also work within the IDE.