openwch/arduino_core_ch32 #213: CH32V103 EEPROM Lib Issues

KenRouKoro, 2025-07-16:
The EEPROM library is not fully successful when running on CH32V103X8T6. Currently, it can successfully read and write the first 2 bytes, but cannot read or write the following 24 bytes, which are always read as 255.

After studying the CH32V103 Flash in depth, I found that the ch32v10x library within the ch32 package in current CH32 platformio and arduino_core_ch32 projects lacks many components compared to the official library. Specifically in the Flash section, the FLASH_ROM_ERASE and FLASH_ROM_WRITE methods are missing. I have tried to create a simulated EEPROM library (not Arduino standard) for reference, which uses the last 128 bytes of the ch32v103. The 128-byte size matches exactly the quick programming size of the ch32v103.

eeprom_emulation.h
```h
#ifndef EEPROM_EMULATION_H
#define EEPROM_EMULATION_H

#include "Arduino.h"
#include "ch32v10x_flash.h"
#include "debug.h"

#ifdef __cplusplus
extern "C" {
#endif

/* EEPROM emulation configuration */
#define EEPROM_SIZE                 128                    // EEPROM size in bytes
#define EEPROM_START_ADDR          ((uint32_t)0x0800FF80) // Flash start address (last 128 bytes)
#define EEPROM_END_ADDR            ((uint32_t)0x08010000) // Flash end address
#define EEPROM_BUFFER_SIZE         EEPROM_SIZE            // Write buffer size

/* Status definitions */
typedef enum
{
    EEPROM_OK = 0,
    EEPROM_ERROR,
    EEPROM_ADDR_OUT_OF_RANGE,
    EEPROM_FLASH_ERROR
} EEPROM_Status;

/* Function declarations */

/**
 * @brief  Initialize EEPROM emulation
 * @param  None
 * @retval EEPROM_Status Operation status
 */
EEPROM_Status EEPROM_Init(void);

/**
 * @brief  Read one byte from EEPROM
 * @param  addr: Address offset (0-127)
 * @retval Read data
 */
uint8_t EEPROM_ReadByte(uint16_t addr);

/**
 * @brief  Read multiple bytes from EEPROM
 * @param  addr: Start address offset (0-127)
 * @param  data: Data buffer pointer
 * @param  len: Read length
 * @retval EEPROM_Status Operation status
 */
EEPROM_Status EEPROM_ReadBytes(uint16_t addr, uint8_t* data, uint16_t len);

/**
 * @brief  Write one byte to write buffer (not immediately to Flash)
 * @param  addr: Address offset (0-127)
 * @param  data: Data to write
 * @retval EEPROM_Status Operation status
 */
EEPROM_Status EEPROM_WriteByteToBuffer(uint16_t addr, uint8_t data);

/**
 * @brief  Write multiple bytes to write buffer (not immediately to Flash)
 * @param  addr: Start address offset (0-127)
 * @param  data: Data buffer pointer
 * @param  len: Write length
[... 318 lines cut ...]
 * @param  None
 * @retval EEPROM_Status Operation status
 */
EEPROM_Status EEPROM_Format(void)
{
    uint16_t i;
    
    if(!eeprom_initialized)
    {
        EEPROM_Init();
    }
    
    // Clear buffer (set to 0xFF)
    for(i = 0; i < EEPROM_SIZE; i++)
    {
        eeprom_buffer[i] = 0xFF;
    }
    
    buffer_dirty = 1;
    
    // Write to Flash immediately
    return EEPROM_Commit();
}

```

KenRouKoro, 2025-07-16:
I pulled the latest example code for ch32v00x, which was updated in March of this year, and I also found the `FLASH_ROM_WRITE` function inside. I think this might be caused by the native library of the current arduino_core_ch32 project not being updated.  
I am currently using ch32v103 in my product, and I have a large number of v103 chips available for testing.
[image]

maxint-rd, 2025-07-16:
With regards to SDK being updated into the Arduino core, I believe you are correct. For the V002/V006 I also saw more recent API code in the SDK than what's included in the current master branch.

Unfortunately at the moment this core doesn't receive much support by WCH. It seems to be very much a community effort by a limited amount of participating members. That's a shame as the CH32 brings good value for money to the Arduino ecosystem, yet interested people may move on to other platforms due to the effort required.

Personally I'm just a hobbyist with no product ambitions. I do have plans to use my chips in my own projects, but I'm afraid I can't do much about the lack of support. For now I try to make things better in my own development environment, sharing my code and experience along the way. 

It would be great if WCH would allow some time by a dedicated developer, but after the efforts of @yangliangquan last April, things appear to have gone quiet again.

maxint-rd, 2025-07-20:
Hello @KenRouKoro - Today I looked further into the issues I had with X033 to use more than 26 option bytes for EEPROM emulation. I had similar issues when testing the V002 and V006. Upon comparing the code for various boards, I noticed that for some boards the flash latency may not be set correctly in the current Arduino core. Looking at the current EVT code in `SystemInit (void)` within system_ch32v00X.c I see this:
   `FLASH->ACTLR = (uint32_t)FLASH_ACTLR_LATENCY_2;`
Some implementations of SetSysClockToXX_HSI(void) in the Arduino core use this:
   `FLASH->ACTLR |= (uint32_t)FLASH_ACTLR_LATENCY_1;`

In the reference manuals for V003/V00X/X03x I saw this recommendation:
```
FLASH_ACTLR
  [1:0] LATENCY[1:0] RW
    FLASH wait status number.
    00: 0 wait (Recommended 0<=SYSCLK<=15MHz);
    01: 1 wait (Recommended 15<SYSCLK<=24MHz);
    10: 2 wait (Recommended 24<SYSCLK<=48MHz);
    11: Reserved.
```
For V103 these recommendations are different (1 wait state for up to 48MHz, 2 wait states for 48-72MHz).  For V20x/V30x I didn't see a latency recommendation.

The [current V10x implementation](https://github.com/openwch/arduino_core_ch32/blob/d76716239cdf8a084a5045c3dfd3151b3f69eeec/system/CH32V10x/USER/system_ch32v10x.c#L565) in the master branch may be incorrect. It has LATENCY_1 for the higher HSI frequencies. The same line in [this EVT source](https://github.com/openwch/ch32v103/blob/f99a84c4c42b6fb676560a9b8b7c737401efe0ad/EVT/EXAM/FLASH/FLASH_Program/User/system_ch32v10x.c#L572) has it changed to LATENCY_2.
That value could be related to the write failures you noticed. 

Note however that there are other differences. For the V002/V006 it seems to use another method to write  option bytes than V003. For the V103 I also see other functions such as FLASH_ProgramHalfWord and some cryptic flag setting [here](https://github.com/openwch/ch32v103/blob/f99a84c4c42b6fb676560a9b8b7c737401efe0ad/EVT/EXAM/SRC/Peripheral/src/ch32v10x_flash.c#L410) in FLASH_ProgramOptionByteData: 
  `*(__IO uint32_t*)0x40022034 = *(__IO uint32_t*)((Address & 0xFFFFFFFC) ^ 0x00001000);`
The current EEPROM library code has no such line.

For now I haven't tried to decipher that line. (My attention is currently focused at getting things better for V002/V006/X033)...

maxint-rd, 2025-07-28:
Interesting indeed. I wonder what it's purpose is. I suspect it to be something related to USB. The CH334U next to it is a 4-port USB 2.0 hub. The 75232G below it is an RS-232 serial driver chip. The X033 has support for USB. Perhaps it's used as a serial to USB converter? I've been using the X033 TSSOP20 for some hobby projects. It can be read/programmed with the WCH LinkE using a two wire interface (SWD/SWIO).

maxint-rd, 2025-12-02:
Correct. In [this commit](https://github.com/openwch/arduino_core_ch32/pull/200/commits/8a07800581bcb37a80065cf45bb636c354a8c03d) I also added support for V002, V006 and X033. From what I've read other series require other changes, but other activities took my attention away and I've not delved further into V20x ,V10x, etc.

I appreciate any contribution by other people. Perhaps one day we can bundle all improvements into a community driven new release...
(Unfortunately I don't have much time to spend on this).

maxint-rd, 2025-12-02:
Hey @fulda1 - Thank you for your efforts to test and improve this. I certainly appreciate _all_ your remarks! Also appreciate the smaller improvements such as better optimization, which I never even thought of and did not take into account. 

It will probably take a while before I get to it, but I intend to incorporate your changes into my code as is. At that time I do want to do some testing myself. (I should have a V203 laying around somewhere, but I never ever got to actually using that chip as my collection of V00x's kept me busy enough...) 

BTW. Most EEPROM code I collected from various sources and kind of stopped thinking as soon as I had something working. Didn't spend much time to find and trim unnecessary code, so I'm glad you took the effort to do that,..

Closing remark: I don't have much experience as a GitHub user, but still would like to see some sort of community driven new release that incorporates all PR that have been submitted. Wishful thinking probably as I don't know how to realize such a thing, but it still would be nice to get a new release someday...