openwch/arduino_core_ch32 #221: What is the best way to configure the external clock?

rtek1000, 2025-09-02:
Hello,

I read that the CH32V203C8T6's internal clock (HSI) can have about a 2.2% tolerance (In 1 hour the clock can reach about 72 seconds of deviation?)

I would like to replace a PIC16F873A and would like to use the 4 MHz crystal present on the board.

What is the best way to configure the external clock, since there is no configuration option in the Tools tab of the Arduino IDE?

rtek1000, 2025-09-02:
I found the file system_ch32v20x.c

(located in .arduino15/packages/WCH/hardware/ch32v/1.0.4/system/CH32V20x/USER/system_ch32v20x.c).

I noticed that if I select HSE 144MHz, it does the multiplication of 18x, that is, 144MHz / 18 is 8MHz.
> SetSysClockTo144_HSE()

If 18x is the maximum multiplication factor. Then with only 4MHz the maximum is 72MHz (4MHz x18).

Then I copied the sub SetSysClockTo72_HSE() and created the sub SetSysClockTo72_HSE_4MHz(). I changed the multiplication factor from 9x to 18x.

Selecting SYSCLK_FREQ_72MHz_HSE_4MHz I now have 72MHz with 4MHz crystal.

`#define SYSCLK_FREQ_72MHz_HSE_4MHz  72000000`

If anyone wants to see it, it is attached.

[system_ch32v20x.c](https://github.com/user-attachments/files/22096664/system_ch32v20x.c)

maxint-rd, 2025-09-02:
> If anyone wants to see it, it is attached.

As you seem to already found out: the v1.0.4 is outdated. The PR's merged into the latest master branch code include selection of clock speed via the IDE menu. Other (unmerged) PR's may be useful too.

In my local installation I also added selection of peripheral devices to the menu. I included those changes in PR #200 (changes made for V003/VM00X/X033, not for other family members since I can't test those).

medtty, 2026-11-07:
Since 1.0.5, pick the external-crystal (HSE) option under Tools → Clock

closed 2026-11-07 by medtty